robotmoose/robotmoose #12: See which pilots are watching robot

olawlor, 2016-06-08:
The frontend should display which pilots are watching your robot while you're driving.

(Low-priority feature request, for AFTER workshop)

mrmoss, 2016-08-03:
I have this working...however it breaks the current implementation of the gruveo dialback feature on the backend. I currently just don't know which pilot to take as the video.

I'm generating a uuid for each pilot and keep track of each of their heartbeats...so this shouldn't be too hard to get to work with the gruveo side of things...

This isn't pushed yet since it breaks the gruveo feature...

Arsh25, 2016-08-03:
How are you handling which pilot gets to drive the robot? I think the easiest way would be to connect back to the pilot driving.

mrmoss, 2016-08-03:
That's not really a thing, any pilot can drive. My current thinking is give
the camera to the oldest pilot. Once that pilot leaves, give the camera to
the next oldest pilot.

Arsh25, 2016-08-03:
Isn't the whole point of having multiple pilots is so that some can watch the robot (sensors) while one commands without having watchers overwrite the commander?

mrmoss, 2016-08-03:
This already works as long as two pilots aren't driving the robot at the
same time (honor system style).

commit 619ee0c8 (2016-08-03): Added multiple pilots and video viewers.
Closes #128.
Closes #12.
closed 2016-08-03 by mrmoss