robotmoose/robotmoose #13: Per-robot chat room

olawlor, 2016-06-08:
To coordinate backend and frontend, having a little chat room visible to everybody would help a lot.

Arsh25, 2016-06-08:
Gruveo has chat inbuilt. We can just use that

Arsh25, 2016-06-08:
It it crazy to use superstar to do this. Send chats to ?chat and just add them to both the frontend and backend

Arsh25, 2016-06-11:
Looking into stuff that's already built. Disqus looks promising,

olawlor, 2016-06-11:
Disqus was quite easy to add to NetRun, and allows a separate "room" per robot.  This is all the HTML it took; you identify the thread by an 'url', in our case robotmoose.com/robots/?robot=2016/uaf/bot

    <div id="disqus_thread"></div>
    <script type="text/javascript">
        /* * * CONFIGURATION VARIABLES: EDIT BEFORE PASTING INTO YOUR WEBPAGE * * */
        var disqus_shortname = "netrun"; // required: replace example with your forum shortname
        var disqus_url = "https://lawlor.cs.uaf.edu/netrun/run?hw='.$hwnum.'";
        (function() {
            var dsq = document.createElement("script"); dsq.type = "text/javascript"; dsq.async = true;
            dsq.src = "//" + disqus_shortname + ".disqus.com/embed.js";
            (document.getElementsByTagName("head")[0] || document.getElementsByTagName("body")[0]).appendChild(dsq);
        })();
    </script>

Arsh25, 2016-06-13:
Looked into Disqus and have a test version on my local system. Big issue is all comments posted as a guest need to be approved.

mrmoss, 2016-06-14:
https://github.com/mrmoss/pychat

Arsh25, 2016-06-14:
I like the idea of using Mike's chat system On hold awaiting discussion with or approval from @olawlor

mrmoss, 2016-06-18:
@Arsh25 I added a push and pop option to superstar, so you can do:
- http://site/superstar/path?push=1234
- http://site/superstar/path?pop=true
- http://site/superstar/path?push=1234&pop=true

Current implementation uses newline delimiters. This way you can do something like:
```
function chat(robot,alias,msg)
{
	superstar_generic(robot,"chat","?get",
		function(data)
		{
			var max_lines=10;
			var uri="?push="+alias+": "+msg;
			if(data.split("\n")>=10)
				uri+="&pop=true";
			superstar_generic(robot,"chat",uri);
		});
}
```

mrmoss, 2016-06-27:
Only implemented on frontend...

mfrancisco1, 2016-06-28:
Looks good.  I really like that the messages stay.  Could we add a shortened timestamp to each messages?

Arsh25, 2016-06-28:
Timestamps are a good idea and can definitely be done. There are two ways we can do this
 ```Pilot 11:41AM: msg``` or ```Pilot 2 Hours ago: msg```. I personally prefer the relative stamp since then we don't have to worry about or clarify timezones.

mrmoss, 2016-06-28:
Sounds good. Let's swap the current storage of line delimited strings to
line delimited json objects in the format:
{handle:"str",message:"str",time:ms_since_epoch}

rykerDial, 2016-06-28:
Relative timestamps are pretty easy with Moment.js. You just store your initial timestamp, then periodically update the time displayed in the chat window with the .fromNow() method. 
http://momentjs.com/

Arsh25, 2016-06-28:
I wouldn't use a JS library for something this simple. The JS ```Date``` object does everything we need.

mrmoss, 2016-06-28:
Javascript already makes this easy:
var diff_seconds=((new Date().getTime())-(new Date(json.time)))/1000;

rykerDial, 2016-06-28:
Moment.js is a wrapper around Date. I just mentioned it because it formats the relative timestamp for you.

mrmoss, 2016-06-28:
console.log(""+new Data());
Tue Jun 28 2016 13:13:07 GMT-0800 (AKDT)

rykerDial, 2016-06-28:
Relative timestamp as in "a few seconds ago", "1 hour ago", "1 day ago", etc.

mrmoss, 2016-06-28:
Oh, that's kind of nice. It's a single file, I'm down.

mrmoss, 2016-06-29:
Still need on backend...

commit 36ecd90f (2016-06-18): Changed superstar from push/pop to append/trim.

Added a simple chat window. Closes #13

Refactored robot ui such that it calls .destroy() on widgets if that exists.

Changed tab "Get Attention" to "Sound" (shorter tab names seem better).
commit b87b522c (2016-06-29): Refactor superstar auth system into calc_auth in superstar.js.
Added auth to append/trim, which in turn adds auth to chat (Closes #93).
Decided NOT to change append/trim system (Closes #94).
Added time to chatter.js (closes #13).
commit 8c7b76f3 (2016-06-29): Added chat to backend (Closes #13).
closed 2016-06-29 by Arsh25