robotmoose/robotmoose #18: Formalized QA/testing and deployment process

FineTralfazz, 2016-06-08:
Normally, we move fast and break things, and that works fine -- until it doesn't, and we're making last-minute changes to try to fix things we broke with our previous last-minute changes. When the infrastructure is broken, we look bad, and when we look bad, people are less likely to work with us. 

Here's a (maybe excessively bureaucratic) policy example I came up with:
* Every commit to master is automatically deployed to test.robotmoose.com using GitHub webhooks.
* Changes which have been tested and verified to work (by someone other than the author) are merged into a production branch.
* Deployments to robotmoose.com from the production branch happen on a set schedule.
	* Production deployments should only happen outside of that schedule in an emergency situation (a critical feature is broken, etc.)
	* Production deployments are the responsibility of one person, and nobody else should do them unless it's both an emergency situation and the regular deployment person is unavailable.
* Superstar's database should be backed up prior to every deployment, so it can be rolled back if needed.

My proposal isn't necessarily the one we should use (for example, I know most of you hate branches), but I think we're reaching a point where we need to have _something_ in place.

olawlor, 2016-06-08:
Yes!   Bucky Frost is scheduled to build an automated test system this summer.  My vision was every night we have a dedicated box that builds an isolated superstar, flashes a Mega, connects a backend, configures the robot, etc.

arcshock, 2016-06-09:
Greetings!

Indeed, I have been getting myself up to speed on the project and investigating a testing process that will best integrate with the team. There is also the possibility for extending the nightly build scripts so that tests can be executed locally to ensure a clean build before committing to remote.

closed 2016-06-22 by FineTralfazz