robotmoose/robotmoose #24: Blockly drag-and-drop programming

olawlor, 2016-06-08:
It'd be awesome to be able to use Google's Blockly to support drag and drop programming.  

We need to keep the ability to switch back and forth to plain JavaScript.

Michaeasmith, 2016-06-10:
Do we want to implement this within the existing state system? Or would this replace it? 

We could create custom "state" blocks that function similarly to the current states (e.g. while loop within function call?).

Arsh25, 2016-06-10:
I think the thing that came up during the meeting was it should be an additional option to the current system. So students can decide if they want to use the block interface or write code. I think some sort of direct translation from the blocks to the robotmoose language would be the easiest way to go.