robotmoose/robotmoose #31: Utility to copy robots

Arsh25, 2016-06-09:
Script or utility to create new robots from already existing robots. Ideally, the user will be able to specify what they want copied.

Eg: Make new KT's from a demo KT and copy over only the config or the config and example code.

computations, 2016-06-09:
Can't the expand_db tool be used for this, or am I not understanding the problem?

Arsh25, 2016-06-09:
The expand_db tool could be used for this, But the idea behind this is to not have to expand the whole db to copy stuff over and then repackage it. This tool would use GET requests and cgi params to copy stuff from one robot to another or maybe even multiple robots.

Arsh25, 2016-06-24:
Reopening to add 

- Ability to recursively go through directories 

- Copy to multiple robots

commit 2046f067 (2016-06-21): Added python util to copy robot keys. Closes #31
commit ce2f9126 (2016-07-04): Now takes a list of robots from a file. Closes #31
closed 2016-07-04 by Arsh25