robotmoose/robotmoose #33: Simplify loading map images

Michaeasmith, 2016-06-10:
Implement easier way to load map images of any scale, tied to a specific robot.

Michaeasmith, 2016-07-20:
I rewrote the code so that it is now possible to use a map of any size. The map creation function takes a JSON of name, width, and height.

There are two questions here that we should probably answer as a group before implementing anything:

1) How should maps be uploaded? Should anyone be able to upload a map image, or should we have a robotmoose page accessible only to teachers/facilitators where images can be uploaded?

2) How should users access the maps? Should they be robot-specific? Should there be some example map images accessible to any robot?

mrmoss, 2016-07-20:
My thoughts:

- I'm still in favor of client side uploads (we host a few "official" maps, any other map is just an image that is uploaded).

- I know that not everyone agrees with this, so I'd be most comfortable with just allowing people with server access to upload images (no user portal at all)...because that's how we end up with awesome new administrators from Romania...

- Ultimately, the best worst-case scenario, we could make a new top level path in superstar for images and store then in binary format. Then we could make a simple web side form to upload images...

Michaeasmith, 2016-07-24:
[image]

[image]

I have successfully landed a roomba sim on the moon.

In other news, I added a client-side upload map image feature. It would be great if someone could test it and/or give feedback about the UI.

closed 2016-08-03 by mrmoss