robotmoose/robotmoose #34: Backend causes apache to lock up

olawlor, 2016-06-10:
Something about having many robots connected causes apache to need a restart.  This is not good for a reliable production system.

It's possible this is related to proxying our comet-style getnext requests.   If this is fixable in the apache config file, we should do that and document it so other people can use the same fix.   If it is not fixable, I'd be tempted to change our getnext implementation, since apache is a sort of default web server, and our software should work with it.

FineTralfazz, 2016-06-10:
I haven't been able to make much progress on fixing Apache. I've found a few people with the same problem online, but no satisfactory answers. One of the suggested solutions seemed to help a little, but it only delayed the hang. Here are a few way we might be able to work around this:

- Switch back to polling.
- Change the getnext implementation from comet to WebSockets, which [Apache has a module dedicated to proxying](https://httpd.apache.org/docs/2.4/mod/mod_proxy_wstunnel.html), so presumably works fine.
- Have clients read a key from Superstar to determine if they should use polling or comet. That would allow administrators who want/need Apache to use the old polling method, while those willing and able to use something else could benefit from the efficiency of comet-style requests.

None of those solutions are great. The first is inefficient, the second would probably require a substantial feature upgrade to Superstar, and the third requires us to maintain two different request methods. I'll keep experimenting with Apache to see if I can make it work with the current implementation, but I'm not very confident that I'll be able to -- some of the threads I found mentioning similar problems with mod_proxy were from >10 years ago.

mrmoss, 2016-06-10:
Short term solution:
Setup a box at superstar.robotmoose.com that only does superstar, then Apache isn't getting in the way.

FineTralfazz, 2016-06-16:
I'm curious to see if fixing the socket leaks (#22) also fixes this.

Arsh25, 2016-06-16:
Test it. Fire up 4-5 roombas and just keep connected for a while

Arsh25, 2016-06-16:
I think 4-5 Arduinos connected to a backend and a pilot should be a good enough test too.

FineTralfazz, 2016-06-19:
I've been testing this over the past two days with the updated backend and trying to reproduce it, and it appears to be fixed. I'm closing the issue for now, if it starts happening again we can reopen and revisit it.

closed 2016-06-19 by FineTralfazz