robotmoose/robotmoose #38: Frontend/Backend password creation interface.

mrmoss, 2016-06-10:
Users need to be able to create passwords on backend (note, we use https, so this shouldn't be too complicated)...

Arsh25, 2016-07-04:
commit a2ac208 added a ?setauth which adds an auth code. Uses is_write_authorized to check if user has parent password before writing the auth

Arsh25, 2016-07-06:
Add ability to add new password and optionally delete an old password.

Change setauth to auth_add and auth_del.

Arsh25, 2016-07-26:
I agree (What!!!!) but I don't see a reason why we need a special denotion for a robot that can't get a password set. I like to think that anyone with sudo (school or / passwords) is trusted enough to not mess with robots. 

We definitely need to create the initial school level passwords but need to give the teachers an easy to use interface to set robot passwords. If we need them to come to us to create a robot or set auth for a robot then that is just another hindrance for them to take over the program without facilitator help. 

Here is my proposed chain of control:
* / password stays with us 
* school level passwords is given to teachers who can use an interface to st robot passwords
* Students get the password to their robot which is set by the teacher

mrmoss, 2016-07-26:
The problem is that we want robots anyone (random people) can drive (aka
write access) but no one can make a password for.

Arsh25, 2016-07-26:
I understand and that is why we should require a higher level password for a password set. That would prevent random people from locking publicly accessible robots and also prevent students from locking teachers and facilitators out from their robots. 

We can also allow password changes for robots unless the password is blank. A higher level password should be able to set a lower password no matter what.

mrmoss, 2016-07-26:
The current implementation allows anyone with a password for their
directory to set a password for their directory and/or subdirectories of
it. The current implementation also does not allow you to set a password
for a directory if it has no password (your last paragraph is exactly how
it is already implemented). With the exception that no one, even root, can
make passwords for directories with blank passwords (to change a password,
it has to already have a password).

Chris was saying it would be nice if we could give students blank robots
and give them passwords. For this, we would need a special read only entry
in the auth file.

In either case, we still need an interface.

mrmoss, 2016-07-26:
Just to clear up that second paragraph:
Give students blank password robots and have them make passwords for them.
To keep random people from making changes to passwordless robots, we'd need
the special read only flag.

Arsh25, 2016-07-26:
Definitely need an interface but that would depend on the API. I know the current implementation. What I am saying is that a higher password should be able to set a password for any subdirectory. A higher password not being able to set a child's password is just weird. 

Should we really allow students to make their own passwords? I'm concerned that students will lock other robots. Teachers should give them a password which they can change.

mrmoss, 2016-07-26:
It can set a password for a child directory, as long as it has a password
already. Thing is, this is kind of nice since a student/teacher can't set a
password for year/school/robot/sensors but they can set a password for
year/school/robot (since we will put a password on it). So if there was an
interface, teachers can change passwords for students (if they forget
them). We can change/create passwords for anyone (since we have root and
access to the server).

It would be nice for teachers to be able to make passwords for new robots
(unless we are going to say we are the only ones who can make robots).

Arsh25, 2016-07-26:
I edited my comment after I noticed parents can set children passwords. 

I agree teachers should be able to make robots in their schools  and set passwords for them. Ideally both creation and password setting will be in the same interface. The interface can be a webpage or somehow integrated into the backend. I prefer it being a webpage. 

TO summarize, I think the current API is good since it prevents miscreants from locking down public robots.

mrmoss, 2016-08-03:
Doing this server side...since we need to make the robots anyways.

closed 2016-08-03 by mrmoss

Arsh25, 2016-08-03:
Not going to re-open this for now. But I think we need a way for users to create robots and passwords. It would help the teachers become more independent from us if they are able to create robots under their schools. Let me know if there was relevant discussion that I missed due to ASRA. If not, this is something we should touch on during scrum this Friday

mrmoss, 2016-08-03:
I'm kind of with Lawlor on this:

We're just an email away (or on-call during the class time).

They usually only need one robot for each student for the entire semester
(all the schools so far have only needed robots at the beginning of the
year).

Otherwise it takes a few minutes for one of us to make robots and give them
passwords (starcp.py and nano/vi auth file).

commit 30d825c0 (2016-07-21): Added change_auth to superstar (server, javascript, and python).
Python now actually supports auth (switch over from python2 to python3 always computed the hash of "").
Closes #38.