robotmoose/robotmoose #44: Touch events on frontend

arcshock, 2016-06-11:


mrmoss, 2016-06-22:
This is actually fairly easy. copy each listener for the following:
onmousedown - ontouchstart
onmousemove - ontouchmove
onmouseup - ontouchend

Mouse based commands start with the passed event, touch based events are one layer deeper, since there can be multiple touch points, so mouse to touch equivalent:
event.pageX - event.touches[X].pageX

For one touchpoint, we should just use the first touch point(0) - which would seem to be a guaranteed index.

DelMitch, 2016-07-13:
Touch events have been added to the Drive UI, but they are untested.

Arsh25, 2016-07-13:
Didn't work in my phone. I believe this is because the parent overwrites the touch events on the child. This goes back to the whole using everslisteners instead of ontouch or onclick