robotmoose/robotmoose #54: Update javascript console.log debug messages to include the source

computations, 2016-06-13:
There are a few debug messages that are in the frontend and maybe the chrome app. For people who are not familiar with the codebase, determining where these debug messages come from is annoying. Just include the file and function that the console.log is coming from to make that faster

Arsh25, 2016-06-13:
In both FireFox and chrome the console shows the file and line from where the message comes from.

computations, 2016-06-13:
guess im just blind

closed 2016-06-13 by computations

mrmoss, 2016-06-13:
It shows the line number of the console.log, but the whole catch errors and print them out has bitten us before on several occasions, I'd personally prefer to find a way to catch, print, and rethrow the errors.