robotmoose/robotmoose #58: create ui widgets that do not wipe during runtime OR ui clear command

mfrancisco1, 2016-06-15:
might be better to make a command for students to use that clears ui if ui gets cluttered
re [https://github.com/robotmoose/robotmoose/issues/21](url)

Arsh25, 2016-06-15:
@mike moss let's talk about this during the developers sessiom tomorrow

Michaeasmith, 2016-06-15:
I cant make it to the developer session today due to doctor appointment, but one option would be to clear the UI as we do now, but give each object an optional "persist" parameter. If persist is defined, the object behaves as all objects used to (checks for duplicate, is not cleared). Objects without persist behave as objects do now, except they won't check for duplicates (I think clearing automatically at each cycle eliminates the need to prevent duplicates, right?).

This way, duplicate labels are fine, except for persistent ones.

mrmoss, 2016-06-17:
Can we get an example use-case for this?