robotmoose/robotmoose #59: Enable duplicate labels on UI

Michaeasmith, 2016-06-15:
Currently, if you try to add a label to the UI with the same string as an existing label, it will do nothing and show no error message.

This is a problem in some cases. For example, if you are using labels to display multiple different numbers (e.g. sensor readings), one will not display if it is the same number as another label.

Buttons and checkboxes behave the same way. Should they be changed as well?

Arsh25, 2016-06-15:
@mikemoss

mrmoss, 2016-06-15:
Yes, this is the same bug that doesn't update the UI buttons and such. The
UI stores a list of UI elements, if the element's value already exists,
then it isn't created.

We should discuss this in detail at the next meeting (Wednesday) and decide
exactly how this should function.

Arsh25, 2016-06-16:
Closed duplicate of #58

FineTralfazz, 2016-07-13:
Should this be closed, or did we decide that we wanted to change how this works?

mrmoss, 2016-07-13:
We defintely need duplicate labels...and I think we settled on a clear
button as well? As far as I know this is still a problem...