robotmoose/robotmoose #64: Backend app should remember the last selected serial device

FineTralfazz, 2016-06-16:


Arsh25, 2016-06-16:
Should it really? The typical usage scenario has only one serial pet anyways. Also what happens if the Arduino is plugged I to a different port, it'll remember a now defunct port.

It currently selected the first one it sees and that is usually ok

FineTralfazz, 2016-06-16:
You're right that our standard Chromebook student setup only has the one serial port, but it's really irritating during development when you have multiple serial devices and the default isn't your Arduino. While fixing #22, I had to change the dropdown every time I reloaded the app. I reloaded the app a lot. It was unpleasant. 

If the remembered port doesn't exist, just select the first one or whatever it is we do now.

mrmoss, 2016-06-16:
It should remember the last serial port used. If the last serial port used isn't plugged in, then they should have to choose it. The serial interface on my OSX machine seems to be broken entirely.