robotmoose/robotmoose #66: Add default change option to checkbox

mfrancisco1, 2016-06-17:
For example:
store.v = false
checkbox("reverse motors", store.v)

Michaeasmith, 2016-06-17:
On git for testing. It requires the parameter to be in quotes. For the variable store.v it accepts "store.v" or "v". So the following lines of code do the same thing:

store.v = false
checkbox("reverse motors", "store.v")

OR

store.v = false
checkbox("reverse motors", "v")

In my implementation, an unchecked checkbox does nothing, and a checked checkbox sets the variable to true (regardless of the variable's previous value). Is this what you were looking for?

I don't know of a simple way to the pass the variable without quotes because Javascript does not pass by reference. If anyone knows a way to do it, let me know.

mrmoss, 2016-06-17:
What happens if you do:
  var v=false
  checkbox("reverse motors", "v")

If it doesn't do v instead of store.v, is it possible to change that?

mrmoss, 2016-06-17:
The only way I can think to get this without quotes is to use a regex to
add quotes before the code eval'd (we start with a string).

Michaeasmith, 2016-06-17:
I forgot about using "var v" in the code editor. That's a local variable separate from any object, so I don't know a way to pass it to a function to modify it. 

But I removed the ability to pass just "v" to modify store.v, to eliminate confusion. So now the syntax is:

store.v = false;
checkbox("reverse motors", "store.v")

Arsh25, 2016-06-17:
A checkbox also seems like a toggle to me. If if I uncheck the box the boolean should go back to false. I don't think that will happen with the current version

Michaeasmith, 2016-06-17:
I debated this with myself for a good three seconds before making it. My thinking was that an unchecked checkbox, like an unpressed button, should do nothing. Unchecked means it could be true or false depending on the code before it, while checked means it is true.

But we could definitely change this so that unchecked = false. Is that more intuitive?

Arsh25, 2016-06-17:
Yeah, that seems more intuitive to me. Looking at Matt's example use case, a checked box means go backwards and an unchecked box could be the opposite aka go forwards.

Michaeasmith, 2016-06-18:
Okay, I changed it so it now sets the value to false if it is unchecked.

Note that this also means that it is possible to give it the name of a variable that does not exist (must be in the form "store.x"), and the variable is created. So it is not necessary to declare or initialize the variable beforehand. It will also overwrite any existing variable (of any type)

checkbox("reverse motors", "store.v")

It's on git now. I'll add instructions to the code help window and API. Please test it.

closed 2016-06-19 by Arsh25