robotmoose/robotmoose #73: Frontend top bar issues.

mrmoss, 2016-06-18:
Frontend top bar doesn't react as expected. When the width of the screen is smaller than the bar, it gets bigger and doorways can go on top of it.

mrmoss, 2016-06-29:
When window starts small the menu moves all the windows about:
[image]

The top bar also does weird things if it starts small and then is enlarged again:
[image]

rykerDial, 2016-06-29:
Ah, thought I fixed that. Where do you define the style for those buttons?

mrmoss, 2016-06-29:
All in robot_menu.js.

rykerDial, 2016-06-29:
As far as I can tell, the buttons were being messed up because doorways set the height of the menu bar to the height of the menu offset, which was not the ideal value when the menu was collapsed. I hacked the correct behavior by setting a static height for the menu bar, but an ideal solution may be to switch from nav-tabs that have their height reduced to a button type.

mrmoss, 2016-06-29:
Feel free to completely swap out that system with whatever you think is
best.

mrmoss, 2016-07-21:
Closing this in lieu of a new menu system in #116.

commit 334d9eeb (2016-06-29): Made frontend top bar collapsible, always stays on top of doorways. Fixes #73
closed 2016-07-21 by mrmoss