robotmoose/robotmoose #74: Switching between saved code can result in overwriting

Michaeasmith, 2016-06-18:
In the code editor, sometimes if you switch between two saved codes and hit "Run" while the dropdown menu is still selected, one of the codes will be overwritten by the other. 

I haven't figured out yet the exact combination of actions to reproduce this every time.

Michaeasmith, 2016-06-25:
I have discovered that this bug can be reproduced without even pressing "Run". Simply switching from one saved code to another and then clicking elsewhere (so that the dropdown is no longer highlighted) can result in overwriting one of the codes. 

If you watch the dropdown, when you switch between codes and then click elsewhere it jumps back and forth between them before settling on one. It seems almost random which one it settles on, but the code itself does not change.

Has anyone else encountered this? It is very, very easy for me to overwrite code, and I have done it multiple times on purpose and by accident.

mrmoss, 2016-06-25:
I'm having a hard time reproducing this bug. This was a known issue at the workshop, but it was fixed.

Where can you reproduce this (robotmoose.com, test.robotmoose.com, localhost)?

Is your robotmoose.com cached by chance?

Michaeasmith, 2016-06-25:
I can reproduce it both on localhost (git freshly pulled, superstar freshly made) and robotmoose.com. I cleared cache on both and it still happens for me. Sometimes it takes about a dozen times switching between 2 codes and clicking away from the dropdown (or hitting Run), and sometimes it happens right away.

If the two code names are different in length, I see almost every time that the name of the other code flashes for an instant when you click Run or click away.

mrmoss, 2016-06-26:
You're running Chrome on Windows 10?

mrmoss, 2016-06-26:
Alright, I think I see it. I can only get it to reproduce if one I start on
Helloworld and go to something else.

Michaeasmith, 2016-06-26:
On Windows 10. I reproduced it just now on Chrome, Firefox, and Edge.

mrmoss, 2016-06-26:
So I just pushed a pretty major change...could I get multiple testers on
this before I close it since it's a pretty important part? @olawlor @arsh25
@Michaeasmith @FineTralfazz

Arsh25, 2016-06-26:
@mrmoss any steps to reproduce the issue?

mrmoss, 2016-06-26:
It shouldn't be reproducible at all. Just try and break it?

FineTralfazz, 2016-06-26:
Latest version is running on test.robotmoose.com. I'll start messing with it.

mrmoss, 2016-06-26:
I'm actually more worried that I broke other things...

FineTralfazz, 2016-06-26:
The code upload feature is gone. Not sure if that was intentional or not.

mrmoss, 2016-06-26:
Do you mean the load file button? I just added that back in if so.

mrmoss, 2016-06-26:
And the load file stuff is broken...fixing now.

mrmoss, 2016-06-26:
Fixed.

Arsh25, 2016-06-26:
Hitting new and then using ```esc``` to quit the dialogue freezes the whole frontend

mrmoss, 2016-06-26:
Not frozen, if you inspect the page, the modal is still there, just invisible...I think that's a bug with the modal class.

mrmoss, 2016-06-26:
@arsh25 Modal stuff should be fixed now.

Arsh25, 2016-06-26:
Verified modal fix

mrmoss, 2016-06-26:
Well, I guess I'll close this...?

closed 2016-06-26 by mrmoss