robotmoose/robotmoose #76: Implement voice commands.

computations, 2016-06-20:
One of the cool things we could do is make a robot listen to commands, and be able to understand a limited set of commands.

computations, 2016-06-20:
One important note for this, when I did some more research, I found that Firefox's WebSpeech API is significantly worse than Chrome's. In particular, MDN's examples don't even work in Firefox, because the API isn't enabled (MDN sites some permissions problems). Even if it got enabled, it would still be worse than Chrome's, due to some of Google's big data "secret sauce". Therefore, this will only be developed for Chrome.

rykerDial, 2016-08-01:
@computations I don't know if you've run into any issues with microphone noise while working on voice control, but I thought I would let you know that the Kinect provides a noise cancelled audio stream. If we decide to integrate the Kinect with the Layla it would be cool if the voice command API could access this audio.