robotmoose/robotmoose #78: Troubleshooting guides

Michaeasmith, 2016-06-21:
Teachers have requested detailed troubleshooting guides for Quizbot and KT.

They should cover all common problems, and they should be written from the perspective of the issues encountered (e.g. "What to do if the robot has no heartbeat").

rykerDial, 2016-06-25:
Created some empty guides on the Google Drive, will start filling them in.
https://drive.google.com/a/alaska.edu/folderview?id=0B49ZOjbHd6YjcmpDQ2N0NEJxTE0&usp=sharing

hmolchan, 2016-07-08:
I've created a section in our wiki for troubleshooting guides. I've started a few but it would be good to collaborate and share this work so it can be added to the wiki.

rykerDial, 2016-07-18:
@hmolchan I've done some work on the troubleshooting guides but I don't know how to edit the wiki. Could you help me get access?

rykerDial, 2016-07-20:
Added a troubleshooting guide for the Quizbot to the git. Will appear on robotmoose after the next pull on the server.

rykerDial, 2016-08-16:
I've added a troubleshooting guide for the KT at https://test.robotmoose.com/kt/troubleshooting_faq/. There are several repeats from the Quizbot troubleshooting guide.