robotmoose/robotmoose #79: build.robotmoose.com subdomain

FineTralfazz, 2016-06-22:
For the automated build/testing project I'm working on, I need to be able to receive GitHub webhooks, which, because of the university firewall, means that I need to be listening on ports 80/443, which means that I need to create an Apache vhost, which is a lot cleaner if it uses a subdomain, so somebody adding a CNAME record for build.robotmoose.com would be nice. 

I don't know who controls our DNS records, but I'm guessing it's either @olawlor or @mrmoss.

Arsh25, 2016-06-22:
Dr Lawlor handles the DNS records. If you prefer a non standard port, we can get that opened from OIT.

olawlor, 2016-06-22:
OK, build.robotmoose.com is now a CNAME for the main server.

closed 2016-06-22 by FineTralfazz