robotmoose/robotmoose #81: Resource leak on frontend

FineTralfazz, 2016-06-26:
Leaving a robot page open on the frontend for a few hours will eventually consume a huge amount of CPU/RAM, making the page, browser, or even entire system slow/unresponsive. I'm not sure what the cause is. I've observed it on FreeBSD w/Firefox, macOS w/Chrome, and Windows w/Chrome. 

Because of how long it takes to happen it's probably not a huge deal, but it's something we should at least be aware of. I'm self-assigning it because I think it's an interesting problem, but if anyone else wants to take a look at it, feel free. It's relatively low-priority and takes hours to reproduce, so I won't be spending much time on it.

Arsh25, 2016-06-26:
This is a known issue I believe. We keep talking about finding a replacement for the whole doorways system. I think looking for an existing system that offers similar functionality is the way to go for this issue.

olawlor, 2016-07-01:
It seems likely to be a JavaScript resource leak, caused by dangling references, not necessarily related to doorways.  There must be tools to examine this sort of thing?

FineTralfazz, 2016-07-01:
I ran the [Chrome heap allocation profiler](https://developers.google.com/web/tools/chrome-devtools/profile/memory-problems/allocation-profiler?hl=en) a few days ago, and it looked like there was something weird happening with Three.js and the robot map. I haven't done enough digging to find out if that really is the source of the problem, but it seems promising.

mrmoss, 2016-07-01:
Is it time to update three.js? Looks like we have version 70 and the newest
is 78. Change log. <https://github.com/mrdoob/three.js/releases>

Michaeasmith, 2016-07-02:
Memory management problems related to map.js or three.js don't surprise me. I noticed some memory spikes when using the map before and after implementing the map image overlay. The spikes seemed to drop back down shortly as the browser did cleanup, but some leaks don't surprise me. 

We should look at where we are calling dispose() with three.js scene objects. Updating three.js would not be a bad idea either.

olawlor, 2016-07-02:
Updating three.js is usually not much benefit but is a lot of pain, because the API randomly changes names and parameters, and you don't see errors from either until the code actually runs (e.g., after you set up a robot with a lidar).  

Calling .dispose() would be a very good idea.

FineTralfazz, 2016-07-13:
I worked on fixing this all day today. Irritatingly, the most significant optimization I tried only required changing 2 lines of code. By drawing the lines on the grid as single lines instead of a bunch of segments, I was able to cut Three.js's memory usage by about 6x on average (60MB -> 10MB, cutting the total page's heap size from ~80MB to ~30MB). 

In my testing it doesn't seem to have broken anything, but I'm not very familiar with the map widget. I'm not sure why it was originally designed to render millions of segments, and there could be a good reason that I'm missing. I would appreciate testing from people who have a better understanding of the system than I do.

mrmoss, 2016-07-13:
I haven't looked at this yet, but are you talking about the actual grid
lines or the lines that are drawn by moving an iCreate around?

mrmoss, 2016-07-13:
If this is just the grid, and the lines still show up via driving, looks good to me. Can't tell for sure until I get back tomorrow morning and get an actual robot.

Michaeasmith, 2016-07-17:
There is definitely a memory leak with the map images. I rapidly switched between the map images and watched the memory usage slowly go up past 1 GB. I'm going to change the map image implementation this week to make it scalable (i.e. map image of any size), so I will also work on its memory usage.

Michaeasmith, 2016-07-20:
I used heap snapshots to look for dangling references and may have found one that comes up at least when you change the map image. THREE.OrbitControls passes a function onKeyDown to Window through an event listener, and this function references the OrbitControls object (which itself references a dom object). Unfortunately, it looks to me like the function passed is unreferenced, so removeEventListener cannot be called on it externally (is that right? someone with better javascript experience can check this).

The attached photos show the heap snapshot and the array of functions belonging to Window - you can see that there is one onKeyDown function for each time the map image was changed.

Does anyone know a way to do cleanup on the event listener functions without re-writing Three's OrbitControls.js?

[image]
[image]

mrmoss, 2016-07-20:
I just pushed a version of orbitcontrols.js with a .cleanup() function that
cleans up the event listener... I did not actually call .cleanup(), as I'm
not quite sure where that should go...

Michaeasmith, 2016-07-20:
Called cleanup() from renderer_t.destroy(). The leak is now gone in heap snapshots, and the memory no longer grows when I switch between map images. 

I don't think this addresses the original issue of CPU/RAM being consumed even when the robot page is idle, though. As far as I can tell this leak was only triggered when changing the map image.

mrmoss, 2016-08-03:
So I don't know if you guys noticed this, but if you zoom in on the thicker lines on the map, they are actually a ton of little lines...

Michaeasmith, 2016-08-03:
I didn't even know you could zoom. How do you zoom?

Maybe we could make it one thick line.

mrmoss, 2016-08-03:
If you're on a robot that doesn't have a roomba, you can do it with the scroll wheel (gif):
[image]

Michaeasmith, 2016-08-03:
Looks like there's an easy way to make that one thick line. I'll do that and see if it gives better performance.

Michaeasmith, 2016-08-03:
Looks like this was a workaround because thicker lines in THREE.LineBasicMaterial do not appear thicker on Windows. I can look into using MeshLine instead of LineBasicMaterial. But using a thicker line did not seem to make a noticeable difference in memory usage, so I'm not sure if it would matter.

FineTralfazz, 2016-08-09:
Closing this, the changes @Michaeasmith and I made have resolved the major performance problems.

closed 2016-08-09 by FineTralfazz