robotmoose/robotmoose #84: Separate giant repo into several small repos?

mrmoss, 2016-06-27:
Should we?

Is there a way to transfer history for certain files to another repo? It'd be nice not to rebase...if possible...

FineTralfazz, 2016-06-27:
This would make the test/deployment stuff that @arcshock and I are working on much easier, and reduce the number of unnecessary merges/rebases. Strong support from me.

FineTralfazz, 2016-06-27:
Also, here's how to split it into multiple repos while maintaining history: https://help.github.com/articles/splitting-a-subfolder-out-into-a-new-repository/

mrmoss, 2016-06-27:
I'm down then...not that my decision pulls any weight... @olawlor?

olawlor, 2016-06-28:
Here's the challenge: we want to make our stuff really easy to access and use, and multiple repos makes it harder for users to collect everything they need.  For example, if you split up superstar and the C++ backend repos, they both depend on some shared network include files, so that'd go into a third repo.  A customer (who inevitably will need a local superstar at some point, for example for testing) then needs to clone and pull three separate repos, and we all enjoy the festival of version mismatch errors when they've inevitably done a pull on two of the three.

The big parts of the project include:

1. Superstar
2. C++ backend
3. Chromeapp backend
4. Web robot piloting frontend (javascript, css, a few index.html files)
5. Other web content, like pictures and tutorials
6. Hardware design files, like LAYLA OpenSCAD/STL files
7. Documentation, like Create2 interface spec

The first four are all tiny, a few megs, and also really tightly coupled.  Adding a feature like chat can touch frontend, superstar, and backend (ideally both C++ and chrome app).  

The last three are bigger (a few hundred megs) and less coupled to each other and the rest of the project.  I could see an argument for splitting those out, maybe into:

- robotmoose/robotmoose: 1-4 above.  Main source code distribution, and bare-bones web page.
- robotmoose/web: remainder of the web content, like tutorials and screenshots and such.
- robotmoose/hardware: circuit designs, debug firmware for new boards, 3D printer CAD part files for quizbot and layla, documentation.

But this is a pretty big change, and will mess up any un-pushed code people have, so we should discuss it at our Friday meeting and make sure everybody's onboard...

FineTralfazz, 2016-06-28:
Yeah, that's a good point. git submodules can help with some of the problem, but they're not a perfect solution.

Arsh25, 2016-07-01:
Final decision is to have 2 repositories:

1.  robotmoose/robotmoose: All software related files 

2. robotmoose/hardware: All hardware related files
..*Schematics and boards
..*Any related firmware 
..*CAD files

mrmoss, 2016-07-01:
So I'm ready to rebase...literally do git rebase master and git push...

mrmoss, 2016-07-01:
In case it wasn't obvious, do not delete ANY history.

olawlor, 2016-07-01:
I'd definitely prefer to not remove history during the shift, but if that's not possible or too hard, we should have a robotmoose/archive with the old commit messages and version history available.

computations, 2016-07-02:
The currently, everything is set up to do the swap. On Wednesday, I will replace the master branch with proposed_master (Doing this Wednesday, is due to the 4th coming up). Any commits to master since todays split will be sorted into either the hardware branch, or the proposed master before swapping. As for the old content, I will put that in a new repo called archive. 

Use this time to tell me that some files should be in one directory or another, or to tell me that I did something wrong, or to generally curse my name. I will get to those comments on Monday, when I am back in town.

FineTralfazz, 2016-07-02:
[I set up a backup repo in case anything goes wrong. Hopefully we don't need it.](https://github.com/FineTralfazz/Chris-Baileys-Home-for-Endangered-Moose)

computations, 2016-07-06:
The swap has been made, people might need to pull the new repo in order to commit any changes

closed 2016-07-08 by Arsh25