robotmoose/robotmoose #85: Sensors issue

mrmoss, 2016-06-27:
Sensors seem to be off by some amount. If you hook up a KT, the floor sensors seem to go into the bumpers sensors. Works fine for a bit, but it messes up after ~10-20 seconds.

Suspect it's between the Arduino and the backend, since it does the same thing on the main site that hasn't been updated in a while...

mrmoss, 2016-06-27:
Do not think it's the backend or frontend, or it's a problem on the C++ backend and the Chromeapp... Want to say it's on the Arduino side...

rykerDial, 2016-06-27:
Floor sensors are still messed up, but bumper sensor is fine. Bumper is changing to 12 when the roomba is lifted up because the wheel drop sensors and the bumper sensors are sent in the same byte, and the firmware and backend are not separating them.

https://cdn-shop.adafruit.com/datasheets/create_2_Open_Interface_Spec.pdf

rykerDial, 2016-06-27:
The floor sensors work when the Roomba is powered off of the battery. When the battery is dead and the roomba is powered by the wall charger, the floor sensors don't work, even when the other sensors do.

closed 2016-06-27 by rykerDial