robotmoose/robotmoose #86: Robotmoose server  switvhed over to Nginx

Arsh25, 2016-06-27:
The robotmoose server suddenly started running Nginx and stopped apache at 5:15 PM today (June 26). No one was on the server as far as @mrmoss and I could tell. We are monitoring the server.

@FineTralfazz : Anything you changed that could do this?

mrmoss, 2016-06-27:
I just disabled the service from auto-starting:
sudo update-rc.d nginx disable

Logs just show nginix starting at 17:07:56. Apache doesn't have anything in
error.log or access.log, other than the absence of connectivity...

FineTralfazz, 2016-06-27:
I installed nginx back when we were having the problems with Apache, but I haven't done anything today. It looks like the server rebooted today and nginx started first.

mrmoss, 2016-06-27:
It was weird. We were testing on the production site, it went down for
about 30 seconds or so until we noticed it. We forced reloaded the page,
nginix showed up.

Arsh25, 2016-06-27:
I'm gonna called this solved if logs show a reboot. Since nginx is disabled
this shouldn't happen again

FineTralfazz, 2016-06-27:
Weird. I guess I'll go ahead and close this, since it shouldn't happen again. Hopefully we don't start having random reboot problems.

closed 2016-06-27 by FineTralfazz