robotmoose/robotmoose #89: Frontend UI Rewrite

FineTralfazz, 2016-06-27:
This came up at the last meeting as something we thought was on the issue tracker, but wasn't. So... is this something we want to do? Is somebody already working on it? Have we made any decisions about it at all?

Arsh25, 2016-06-27:
No one is working on this yet. And no decision has been made about it. First step is we need to find some library for franeowke that gives similar functionality as doorways, is easy to customize and reliable.

mrmoss, 2016-06-27:
I like c9's interface...not sure if it would work for us though.

FineTralfazz, 2016-06-28:
Whatever the UI ends up looking like, I'd be in favor of building it on top of Angular or Ember or some other JS framework so we don't end up reinventing the wheel with a bunch of weird custom stuff and writing a bunch of boilerplate DOM binding code. I might be alone in that, though.

mrmoss, 2016-06-28:
I've uses Angular a bit in app development (Ionic uses it). Could you mock
up an example Angular/Ember frontend? Doesn't have to be 100% functional,
or even 10% functional, just to give us an idea of how/what it can do for
us? I'm completely open to entirely changing the frontend.

FineTralfazz, 2016-06-29:
I'll throw something together and see how well it works.

Arsh25, 2016-06-29:
Apparently Google came up with for Gmail https://material.angularjs.org/latest/

FineTralfazz, 2016-06-29:
I've spent the last few hours experimenting with Ember. It's pretty nice, but it's very opinionated about how APIs format their data (it basically demands the [JSON API spec](http://jsonapi.org/)). Normally that wouldn't be a problem, and we could write custom adapters/serializers to make it work, but the way we're currently formatting things with Superstar is so different that writing adapters and serializers to work with it is a huge pain. Even with a semi-working one I wrote, it feels so hacky that we might just be replacing one set of weirdness for another. 

Angular doesn't have that problem, but I started with Ember because Angular is currently going through a significant compatibility-breaking transition from v1 to v2. The tools/docs aren't very good yet, and the community seems pretty strongly divided. It's hard to tell how it's going to shake out. It might take a few more months, or it might be a Python2/3 situation where it takes years. Angular2 also _strongly_ favors TypeScript, which I'm not 100% opposed to, but like the Ember adapters it's another layer of weirdness. 

I really like Ember -- it's well-documented, clean, and relatively easy to use. But creating the bridge layer between its models the Superstar format is probably more trouble than it's worth. Given more time/resources I would replace Superstar with a normal API/database, which wouldn't be that much work using something like Rails; but at that point we're basically rewriting everything, which is a tough sell.

mrmoss, 2016-06-30:
What would be the ideal database? MongoDB? Superstar is a little limiting...there's been several situations where a real database would have solved our problems (authentication, chat, etc...). It's probably about time to migrate to a real database.

Arsh25, 2016-06-30:
Had a quick look at mongodb. It does introduce a lot more dependencies but it seems like the ideal DB for our use. There's also firebase which is a SaaS database by Google. O have no idea on what the limitations on the free plan for that are. 

I think we should add this to the Friday agenda and discuss it there.

mrmoss, 2016-06-30:
To be honest, 99% sure every server will be unix based...so dependencies aren't really an issue.

I'm thinking MongoDB is the way to go (Firebase looks to still be in development). Looks to be as easy as adding an apt repo and apt-get install mongodb-org. Only uses a single tcp port 27017. As always Digitalocean has a great guide on how to install/configure it.

Arsh25, 2016-06-30:
Again I haven't looked into this too much but I was talking about
dependencies for interacting with MongoDB with C++
https://github.com/mongodb/mongo-cxx-driver

mrmoss, 2016-06-30:
Thing is MongoDB is not meant to be public facing, so we'd have to use a web interface (Mongoose...which is Node.js based).

Why are all of these things so complicated? We'd need to have Mongoose, MongoDB, and Ember?

Arsh25, 2016-07-01:
I am ok with this, we'll need something to interface with MongoDB anyways.
Mongoose seems nice since that means our entire stack is JavaScript.

As to why it has to be like that, well welcome to real dull stack
development

FineTralfazz, 2016-07-01:
The database choice is less important than wrapping it in a standard JSON API. Honestly, I would probably just use MySQL or Postgres.

Arsh25, 2016-07-01:
The beauty with MongoDB is it does JSON by default. I'm sure it follows all
the standards.

mrmoss, 2016-07-01:
Would we need to build that wrapper or is there something you'd recommend?

FineTralfazz, 2016-07-01:
We'd need to build it either way. Using Rails with something like [ActiveModelSerializers](https://github.com/rails-api/active_model_serializers) makes it fairly easy, though.

Arsh25, 2016-07-01:
So we just added layer. Is this what the new schema would look like DB > something that interacts with the DB > Serializer > Frontend ?

FineTralfazz, 2016-07-01:
Essentially, except that `something that interacts with the DB` and `serializer` are the same thing.

mrmoss, 2016-07-01:
If that's the case, I'd rather do something like MySQL and Python...or build a Python interface to MongoDB. I know RoR is used a lot for this type of thing...but I associate RoR with drowning kittens...not sure why...

Arsh25, 2016-07-01:
I agree with Mike RoR instils fear into my mind. I would still suggest MongoDB because we can serialize the data once and store it in the correct instead of fetching raw data and serializing it every time. Don't know if this will help but the idea of having a full RDMS seems overkill. 

From a server point if view, the big thing superstar does not do is follow the response codes specified by the specification. If we were to meet the specification, it seems like we would have to rearrange all our objects

FineTralfazz, 2016-07-01:
We can prototype multiple options -- if we're going to make a change this significant, it might be a good idea to try a few different solutions, compare them, and decide on the one we thing works best. 

Also, you guys should try Rails. Seriously. It's so good.

mrmoss, 2016-07-01:
I'd almost prefer a Javascript interface, then we would only have one code
base for everything.

olawlor, 2016-07-01:
Rewriting the frontend UI, and rewriting the network communication backend, should be two separate issues.  If they're coupled, that's not a good sign for the UI framework.

One thing I never liked about databases is the installation hassle of setting up the tables.  Right now, a user that wants a standalone superstar (e.g., because they have no useful network in their classroom) can just run one binary and they're done.  Setting up a database isn't that bad, but it's pretty bad, especially on non-UNIX systems.  

Recall the hassle with getting Indiana's original websocket / java / mysql stack running...

Bare minimum, for a superstar replacement I'd like to see:

- Standalone setup script that is guaranteed to work on Linux, OS X, Windows.
- Similar or reduced network traffic (Anything sending XML or other bloated crap on the wire is out.  Anything doing separate TCP handshakes for every data exchange is out.)
- Comet support (I think this is key to getting good performance from always-on LAYLAs; polling doesn't scale).

mrmoss, 2016-07-19:
In the process of moving the [doorwaysv4](https://nullify.cc/doorwaysv4/). This is a pretty major integration...if anyone is interested in pair-programming with me, I'll be in Tuesday/Wednesday (it's nice to have someone watching for errors).

The new doorways api is much easier to work with...but this also means that the original file is more or less useless (since it has a kind-of convoluted chain of loads/saves).

I figure superstarv2 might as well be integrated more or less at the same time (since that interface is pretty different as well).

mrmoss, 2016-07-21:
See #116 for details on why this is being closed...

closed 2016-07-21 by mrmoss