robotmoose/robotmoose #91: meta-robot capability

mfrancisco1, 2016-06-28:
a robot page that can control two or more backends
backends can be of traditional type or can be camera-only and arduino-only backends

this would become something we would want once Android and iOS backends are established

benefits: another way student teams can collaborate, can make more complex robots,

Arsh25, 2016-06-28:
The way the system works, 2 backends can have the same name and be commanded together however sensor data will have a race condition and make no sense. 

Do we want to b able to command two different robots simultaneously or do we want to have a separate drive UI for each robot? This question can be expanded for all the other elements (doorways) on the frontend.

mrmoss, 2016-06-28:
It would be interesting if we could configure a "Moose" object, (which is
just a robot name). Backends could theoretically control other backends
this way.

olawlor, 2016-07-01:
I always wanted a way for the frontend to read sensor values from other robots, like:
   robot("2016/uaf/addon").sensors.analog[0];

It's not clear how to generalize this to commanding other robots, in particular for the case where multiple pilots are sending conflicting commands (but is this even an important case?).

Arsh25, 2016-07-01:
An easy way I can see of allowing multiple pilots to command a robot is to have a "lock". The lock has to be willingly given up by the holder to before another pilot can drive the robot. Once the lock is released it becomes available to be grabbed by another pilot. If we want to guarantee that the lock will be released then each lock could have set time limit.

FineTralfazz, 2016-07-01:
Wouldn't only writing pilot commands on change (instead of every x ms) work too? Seems easier than dealing with locks.

Arsh25, 2016-07-01:
Yes that would be easier. But what happens if both pilots are writing at the same time. There's really no issue if only one pilot is writing at a time