robotmoose/robotmoose #92: Test server javascript is broken

rykerDial, 2016-06-28:
Somewhere in the test server javascript code the code is broken, because most of the UI elements no longer function. I rolled back my local repo and tested until it was fixed. This commit https://github.com/robotmoose/robotmoose/commit/9d7402bdea60757864c94a62efd9ed58bfa8d9bb seems to be the culprit. @mrmoss

Arsh25, 2016-06-28:
That commit was when the new multiple superstar paths syntax was introduced. My guess is the doorways don;t work because of something that the new networking protocol messed up.

mrmoss, 2016-06-28:
Nope, this is because I commented out the try/catches in superstar.js...so
something is requesting something that is not a json object and actually
throwing instead of being saved by the superstar function...

Arsh25, 2016-06-28:
Seems to be some malformed json in the new multiple set and get system. 

[image]

mrmoss, 2016-06-28:
I'm not getting any errors...

Arsh25, 2016-06-28:
They are no errors in the console, the requests are coming back with a malformed JSON object as shown above

mrmoss, 2016-06-28:
Malformed JSON objects in Chrome show up in the console (not as an error, but console.log because superstar.js saves it).

The whole "multiple get" thing isn't new. It's been there for a long time (I added it to the frontend, but the backend has had it for a long time).

I did move chat to the multiple get request thing. The chat system isn't using JSON objects, so it isn't decoding correctly. If the chat history was empty, the multiple get system makes it an empty object. I was testing on a stock robot that had no chat history...hence why it worked.

commit f626bd14 (2016-06-28): Superstar.js now does JSON parsing (instead of relying on the client to do it). Robot_network.js is now pulling all the objects from the multiple request. Closes #92.
closed 2016-06-28 by mrmoss