robotmoose/robotmoose #100: Superstarv2 Unit Tests

mrmoss, 2016-07-09:


mrmoss, 2016-07-10:
It's probably better for someone other than me to do these...but I can at least start them out...

Arsh25, 2016-07-10:
This is something I could do. [catch](https://github.com/philsquared/Catch) is something easy to use unless you want to rill your own system. 

Going to assign myself too

mrmoss, 2016-07-10:
I've always found using Catch for any actual test bed rather difficult to use. It makes adding tests easy...but it makes debugging and reading why tests fail difficult...which is the opposite way of every testing lecture I've been to.

I think this is literally a vector of a test struct that has the thing you want to test in it and what it is expected to output (since everything is a get/set).

I'd also like to add in a separate timing analysis testing program.

Right now there's a blank unit_tests.cpp that's been in there since the beginning of development...I'll start adding in some simple test cases.

The majority of the help I'm seeking is more test cases rather then writing the code for them. I think I've got the majority of the "weird requests" and technicalities of using arrays/objects as a path...but I'm not sure.

mrmoss, 2016-08-03:
I wanna say this is done yes?

closed 2016-08-03 by mrmoss