robotmoose/robotmoose #104: Superstarv2 migration.

mrmoss, 2016-07-10:
As soon as testing and such is done, we need to migrate over to the new version (assuming we still want it), mainly since it isn't a "drop-in" change over since the new backend has a very different system...transferring old databases and such might be tricky.

Arsh25, 2016-07-10:
For DB transfer I'm thinking we could use Ben's expand utility and parse the directories produced into a the new format or create a utility that copies an old robot into the new format. In either we would need a new utility.

mrmoss, 2016-07-10:
I'm planning to make the old database format loadable by the new
version...but I'd like to start storing the database as a JSON object (so
just rewrite the database as a JSON object from then on).  This is issue is
more or less just a formality...I actually think the migration will be
pretty simple.

mrmoss, 2016-07-21:
Just moved test.robotmoose.com to superstarv2, please take a look and see what's broken.

mrmoss, 2016-07-21:
All backends are broken...

Michaeasmith, 2016-07-21:
Two things I just noticed from home: 

Did a git pull, ran local superstar, and the page gives an error when I click on "Robots": 
VM1626:1 Uncaught SyntaxError: Unexpected token < in JSON at position 0
xmlhttp.onreadystatechange @ superstar.js:149

Secondly, on test.robotmoose it looks like simulations are no longer supported with the new config_editor.js, robot_network.js, and maybe other files, but I can fix that.

mrmoss, 2016-07-21:
I'm actually going to revert to the old version...I don't want to leave this broken...

mrmoss, 2016-07-21:
I'm guessing your error was a caching/old version issue (new superstar expects JSON), were you running superstarv2?

I moved this to a new branch: superstarv2_migration

Michaeasmith, 2016-07-21:
Oh, nope, I was running the old superstar. I tried running it before even looking at the files, and I had assumed you had just replaced superstar.cpp. I'll play around with the new branch.

closed 2016-08-03 by mrmoss