robotmoose/robotmoose #105: Superstarstarv2 copy robots util

Arsh25, 2016-07-12:
Make a utility to copy robots for Superstarv2.

mrmoss, 2016-07-22:
Blind guess...but I think in python this is:
```
#usage: ./copy_robot SUPERSTAR FROM TO
import getpass
import superstar
import sys
if __name__=="__main__":
  if len(sys.argv)!=4:
    print("Usage: ./copy_robot SUPERSTAR FROM TO")
    exit(1)
  if sys.argv[2].count("/")!=2:
    print("Invalid FROM robot path.")
    print("Usage: ./copy_robot SUPERSTAR FROM TO")
    exit(1)
  if sys.argv[3].count("/")!=2:
    print("Invalid TO robot path.")
    print("Usage: ./copy_robot SUPERSTAR FROM TO")
    exit(1)
  auth=getpass.getpass(prompt='Enter auth:  ')  
  ss=new superstar_t(sys.argv[1])
  def error(error):
    print(str("Error("+str(error.code)+") - "+error.message))
    exit(1)
  def success(result):
    print("Success!")
  def do_copy(data):
    global auth
    global ss
    ss.set(sys.argv[3],auth,success,error)
  ss.get(sys.argv[2],do_copy,error)
```

mrmoss, 2016-07-22:
Someone want to test this (I've tested...but it seems like someone else should...)?

commit 0de449b0 (2016-07-22): Added a copy robot script...
Closes #105.
closed 2016-08-03 by mrmoss