robotmoose/robotmoose #106: Frontend simulation testing

Michaeasmith, 2016-07-14:
I finished the frontend simulator, but it would be nice if someone could test it and take a look at the code, to make sure it is working well and I didn't break anything else in the frontend.

The frontend simulator stores the experiments, configuration, and gui data in localStorage. For a create2, it simulates movement using the same code as the robot_2wd simulator we used for the arduino code simulations.

By the way, what is the right procedure here for committing something to close an issue when the commit needs testing? Is it right to close the issue and make a new issue for testing, as I did here, or is it better to leave the issue open and request testing in the comments?

Arsh25, 2016-07-14:
This is a good question. I think the cleanest way to do it would be using
Milestones <https://guides.github.com/features/issues/#filtering>. My
proposed workflow:

   1. Work on an issue
   2. Test it, commit to repo and close the issue
   3. Assign the "Testing" Milestone
   4. People looking to test can search by this Milestone, test stuff and
   comment on the issue they are testing.

This is something that should definitely be standardized and worth talking
about on Friday.

mrmoss, 2016-08-03:
@Michaeasmith You want to test this again? It seems to work for me...but you may know some edge cases I'm not thinking of...

Michaeasmith, 2016-08-03:
It's working for you? On test.robotmoose when running a simulation it won't let me select a device on the configure tab.

Michaeasmith, 2016-08-03:
Just pushed a fix. Simulation seems to be working again for me on test.robotmoose. It would be good if someone else can try it too.

mrmoss, 2016-08-03:
Caching?! Looks good now.

closed 2016-08-03 by mrmoss