robotmoose/robotmoose #108: Generic Superstar Test Script

FineTralfazz, 2016-07-15:
It was decided at today's meeting that there should be a script to test Superstar JSON-RPC requests, both to make sure that we don't accidentally break the C++ implementation and to test alternate implementations for feature parity.

FineTralfazz, 2016-07-21:
This is partially done. I need more test cases though, and I'm waiting to implement auth testing until #38's API is implemented.

mrmoss, 2016-07-21:
Just added auth interface.

Notes:
- Can only change passwords, cant make them (so if a path has an empty password or it's not in the auth file, then this doesn't do anything.
- Breaks multiple passwords in auth file (was more of a surprise feature anyways).
- Password restrictions: Printable non-whitespace characters with length of 8+.

FineTralfazz, 2016-07-22:
Is there an architectural reason it can't create passwords, or is it just an implementation limitation?

mrmoss, 2016-07-22:
We want to have robots that are open to anyone. If there isn't a password,
and you could create one, someone could create a password for one of those
robots.

FineTralfazz, 2016-07-26:
It would be nice to change that somehow, so that users could create their own passwords from the frontend and/or backend, without relying on a person with server access. Possibly a special flag for publicly-accessible robots which allows unauthenticated set/get/etc. calls, but doesn't allow creating or changing passwords, or something? I don't know.

Arsh25, 2016-07-26:
This is something that needs to be implemented and I think can be done if the person has a higher password (like the password for the school). There is  an open issue (#38) which should not have been closed since it specified that we needed a way to create a password from the front or backend and not via a special script. 

Providing an easy GUI way to set a password if you have a higher password should not be an issue since I hoe that a trusted person with a school level password is not malicious.

FineTralfazz, 2016-07-26:
I'm going to close this issue since it's essentially done. There are a few more tests which should be added, like auth and getNext, but the core functionality is there. 

Re auth: I'm not sure if we should create a new issue or reopen #38. I'll let you decide.

closed 2016-07-26 by FineTralfazz

mrmoss, 2016-07-26:
I talked to Lawlor about this. He said he's ok with us creating the initial
password for the users via the server.

But you are right, we could simply say if you put a * as a password in
auth, it's a robot with a blank password that can't be set. Be easy enough
to implement. Then you can use the interface we already have (in terms of
the API call, the actual modal/inputs we don't have at all).