robotmoose/robotmoose #115: Superstarv2 Backend Integration

mrmoss, 2016-07-21:


mrmoss, 2016-07-22:
I'd really like to settle on an interface for everything before this is written. It feels like superstar should only do superstar stuff (take a path and a value, not a robot object). On that note, how about something like:

```
//Our robot object looks like:
{
  superstar:URL,
  year:YEAR,
  school:STRING,
  name:STRING,
  auth:STRING
}

//Tries to connect to the given robot object.
//  Would validate year/school/name and auth.
robot.connect(robot_object,onsuccess,onfail)

//Do all the standard things we support in batch.
//  Do auto flushing in background?
//  Alternatively we could do the single arg based version of this...
//    Aka {path:STR}, {path:STR,value:ANY}, {path:STR},
//           {path:STR,value:ANY,length:INT}, {path:STR,value:STR}
robot.get(path,onsuccess,onfail)
robot.set(path,value,onsuccess,onfail)
robot.sub(path,onsuccess,onfail)
robot.push(path,value,length,onsuccess,onfail)
robot.change_auth(path,value,onsuccess,onfail)

//Get next which is not part of the batch operations.
//  We'd need to add comet support to superstarv2.
//  Again, could do {path:STR}
robot.get_next(path,onsuccess,onfail)
```

We also need to be able to get to superstar to begin with (to be able to select a robot). For that, I kind of just want to have a global superstar that anyone can use:
```
superstar.connect("https://robotmoose.com")
superstar.sub("/robots",function(years)
{
    print(years)
})
```
Thoughts? @Arsh25 @olawlor @FineTralfazz

Arsh25, 2016-07-24:
I don't know if this was discussed at the meeting and a decision was made. Ignore this if a decision has been made (Or not).

I definitely agree that Superstar should only take a path and value and should not be changed to take a robot object. 

I like the suggested object. This seems to be pretty standard way of doing JS stuff. The only thing I disagree with is having superstar being a property of robot. A robot having a superstar does not seem right, it almost seems like a superstar should have a list of robots.

mrmoss, 2016-07-24:
A superstar object is simply an object that can be used to communicate to a
superstar (more of a client).

Arsh25, 2016-07-24:
Exactly my point. I was referencing your suggestion of 
`robot.superstar.sub("/robots",function(years)
{
    print(years)
})
robot.superstar.flush()
`

This suggest that superstar is a property of robot. If I understand this correctly, it will store all robots in a superstar into the robot object. It makes more sense to me that a Superstar had robots, so we would have a superstar object that has a method to get all robots and each robot would be a robot object. So the above would be 

`var robots = superstar.sub("/robots",function(year)
{
    //Do stuff to append a robot to a list of robots with robot.year=year here 
})
superstar.flush()
`

mrmoss, 2016-07-24:
We want the ability to connect to two different robots that may be on
separate superstars, so the one global superstar will more than likely
hinder that feature in the future. Each robot kind of needs its own
superstar internally anyways (for the backend side, the frontend is always
going to be limited to a single superstar on it's side, only way to do this
is to proxy commands through the backend).

Since we want to start combining all the software to a single API, the
decision was made not to have the global superstar since that's going to
work on the backend and the frontend (the global superstar would work
flawlessly on the frontend, but would need some tweaks on the backend for
multi-robot support).

closed 2016-08-03 by mrmoss