robotmoose/robotmoose #119: Reset map position?

Michaeasmith, 2016-07-27:
It would be nice to be able to reset the 3D roomba to the middle of the map when the map image is changed. More generally, it would be nice to be able to move the roomba to any location on the map. This is especially important since some maps can be larger than others, so switching maps after driving can result in the roomba being lost in space.

But given that the sensors.location is shared with the backend, what is the best way to implement this?

1) We could send the backend new values for location sensor values. But this could disrupt any algorithms running on the backend (e.g. localization algorithms), and I don't know if it is correct to change sensor data from the frontend. 

2) The map could use virtual, phase-shifted location values. The frontend could store both the real sensor values and the virtual sensor values, and we could choose how to make these sets of values visible in the UI and accessible from the code editor (e.g. "location", "map_location"). But this might be adding unnecessary complexity.

3) We could leave the sensor data alone and phase-shift the map coordinates. But this is only an option if we leave out the ability to reset the angle of the roomba. Otherwise, the x and y sensor locations may have no correspondence with the x and y axes of the map, which would be entirely too confusing for the user (and would make programming using location data almost impossible).

Which is the best option? Or is there another?

mrmoss, 2016-07-27:
What if we just had a reset position button? If we used SLAM, the position
would auto update back to where it was after the reset. If we used ARUCO,
the position would also auto update back to where it was.

Arsh25, 2016-07-27:
The frontend should NEVER set sensor values. I think the best thing to do for now (without any information from Neato) is to reset the roomba to a fixed location on the map (center?) and assume that to be the starting positions and then update location as we are currently. 

If we get localization via Neato (or Aruco) then we could update the virtual location the where the robot thinks it is (assuming we are actually in the location represented by the map).

mrmoss, 2016-07-27:
Why shouldn't the frontend be able to set sensor values?

The robot has no idea where it is to start out with. A human has to tell
the robot where it is starting (unless it sees an ID with ARUCO). Other
than that case, human interaction serves as a sensor.

If someone places the robot in a room, and, as a pilot, you see a landmark
that matches on the map that doesn't match up with your current location
via video, why shouldn't you be able to say "robot is actually over near
this landmark" since you can see it?

Arsh25, 2016-07-27:
The Frontend should not be able to set a sensor value, so pilot should not be able to say that the sensor.bumper=1 for example. But yes, the way you described is totally ok, a pilot should be able to tell the robot that hey you are on this location in the map. Setting starting location is ok but frontend should not be able to reset the location sensor (location.x=200 but location.x=0 is ok). I believe this is what Micah means by phase shifted location values.

mrmoss, 2016-07-27:
So if you are driving around, and notice the robot is completely shifted by
a room or something because it was not placed correctly in the starting
area, the pilot should have no control over resetting that? Again, if the
pilot can see exactly where they are, why not let them fix it instead of
waiting for someone else to fix it for them?

If a student is constantly testing a pre-programmed movement (like the
robot glasses exercise), the pilot shouldn't be able to reset that position
over and over again? We can say add it the backend, but it is kind of
inconvenient since the map is not currently visible...

Arsh25, 2016-07-27:
We're saying the same thing. If the pilot wants to reset it they move the robot on the map to where it actually is and reset the location. All locations are relative to a zero anyways( current implementation). 

So yes, the pilot should have control over resetting it's location.

Michaeasmith, 2016-07-27:
If no one else has objections, I will go ahead and implement a reset location button that will reset the sensor.location values to 0. 

Later, I would like to implement the ability to view the map image and click on any spot to set the location.