robotmoose/robotmoose #120: Switch camera button on backend.

mrmoss, 2016-07-30:
Can't switch cameras for gruveo on backend.

rykerDial, 2016-08-03:
Been doing some research on this. We can get a list of available cameras with navigator.mediaDevices.enumerateDevices(), but there doesn't seem to be a way to change the camera source without direct access to the video object.

mrmoss, 2016-08-03:
We are hiding the permissions window, maybe we should just let it popup so you can choose the camera?

rykerDial, 2016-08-03:
Probably the best option, though I'm not sure how to do it. When you remove the permission request event listeners there is no popup, just denies permission automatically.

Arsh25, 2016-08-03:
We aren't hiding the window. The app has permission to access the camera and microphone via manifest.json. Maybe we can capture the event and do something with it instead of just allowing it?

mrmoss, 2016-08-03:
If you get rid of that, doesn't it ask for permission?

Arsh25, 2016-08-03:
No it denies the request if you do not capture the event

rykerDial, 2016-08-03:
You can have this one Arsh :)

rykerDial, 2016-08-03:
We may just have to tell people to change their camera the standard way: Settings -> Show Advanced Settings -> Privacy (Content settings) -> Camera (Choose camera from dropdown)

rykerDial, 2016-08-03:
If you decide to pursue creating a button, here is some code I wrote to make a dropdown of cameras:

```javascript
function media_device_selector_t(div) {
    if(!div) return null;

    this.div = div;
    this.el = new_div(this.div);

    this.dropdown = new dropdown_t(this.el, null);
    this.dropdown.set_width("100%");
    var _this = this;
};

media_device_selector_t.prototype.update_list_m = function() {
    var _this = this;
    navigator.mediaDevices.enumerateDevices()
        .then(function(devices) {
            var videoinput_labels = [];
            devices.forEach(function(device) {
                if(device.kind === "videoinput") {
                    videoinput_labels.push(device.label);
                }
            });
            _this.dropdown.build(videoinput_labels, null);
    });
};
```

Arsh25, 2016-08-04:
Just need to figure out how to use the selected camera (See commit 6ff8bb6ce8e6802af5c6368e751c26cc7103466c) If someone knows how to do that go ahead. I have this crazy idea of listening for the permission request and sending it to a notification and see what that does.

Arsh25, 2016-08-06:
As discussed during the weekly sxeum, this issue is low priority and temporarily abandoned since there isn't a way to do this. If someone finds a way, please reopen and work on it. Closing issue.

closed 2016-08-06 by Arsh25