robotmoose/robotmoose #122: Serial dropdown on backend doesn't auto load serial devices.

mrmoss, 2016-07-30:
Was seen today during a walkthrough, serial devices do not seem to update on the backend on Windows (haven't tested on other systems).

mrmoss, 2016-08-03:
I think this is actually a dropdown issue not a name.js issue...

rykerDial, 2016-08-03:
If your list of serial devices changes while the serial dropdown element is selected, the dropdown won't refresh until you click on something else.

Arsh25, 2016-08-03:
Notice same behaviour for all dropdowns (changing SS does not update schools if dropdown is open) this leads me to believe it is a dropdown_t thing. Should we even worrying about it not updating when open?

mrmoss, 2016-08-03:
Yes, because then you can't ever select anything...

mrmoss, 2016-08-03:
@Arsh25, so I noticed if you replace the dropdown.js in the app with the one on the site...it works...making a softlink...

closed 2016-08-03 by mrmoss

Arsh25, 2016-08-03:
I'm not assigned to this so thx

mrmoss, 2016-08-03:
Oh, I thought you took it?