robotmoose/robotmoose #123: Subsystem Integration with Robotmoose

rykerDial, 2016-08-02:
As we continue to develop subsystems such as the Neato and the Kinect, we are going to need a method for integrating them with Robotmoose. The current stack assumes everything goes through the Arduino, so we will need a way to expand this. I had a few ideas:

1. Subsystem programs run as separate programs or as children processes of the chrome app. They pipe or otherwise transfer their data to connection.js which sends it to Superstar.

2. Subsystems run on their own server(s) and send their data directly to Superstar. I like this method because some subsystems (like the Kinect) have drivers and other dependencies that need to be installed, and it would make them independent of the main app.

If there is some better or standard way of doing this, please let me know. We should come up with a standard way of integrating subsystems.

Arsh25, 2016-08-02:
Your first suggestion is basically the "standard" way to do it. All subsystems are clients (Neato and Kinect stuff) for superstar. They do their own processing (separate from the chromeapp) and send their data to superstar. The data will be sent to superstar via the superstar API and the API has been implemented in JavaScript, Python and C++ (Maybe ?). 

I can see some argument for integrating the Neato and Kinect into Chromeapp but I think that the ChromeApp should be the bare bone basic robot that anyone can setup without having to worry about dependencies.

mrmoss, 2016-08-02:
Neato should be implemented in Chromeapp (since this is simple serial
processing, Aven is currently working on this).

Kinect has to be separate, since the only way (currently) to get it to work
in a chromeapp is a browser plugin.

mrmoss, 2016-08-02:
As for the superstar c++ client, it looks like Ben made one. Although it
uses a separate hash library (it be good to swap it over to using
auth.cpp/hpp and picosha and link the files if possible). It may also be
good to rename superstar.cpp/hpp to superstar_server.cpp/hpp (this way the
interface to superstar is the same in every language minus the syntax
differences).

I actually haven't looked at the implementation at all, but it'd also be
good if it used the same callback system as well (vector of request structs
containing function pointers).

rykerDial, 2016-08-16:
This part of this problem is practically over with. Superstar clients exist for C++, Python, and Javascript. Subsystems will exist as separate programs when it makes sense to do so. #131 and #137 follow up this issue.

closed 2016-08-16 by rykerDial