robotmoose/robotmoose #124: Backend auth error

mrmoss, 2016-08-03:
Backend opens an auth error window with nothing really in it (I made this a long time ago hoping someone would finish it...).

mrmoss, 2016-08-03:
Should this even be a popup?

Arsh25, 2016-08-03:
I think a popup is good. Makes it really hard to miss. Maybe some annoying music too (Pls don't do that)

mrmoss, 2016-08-03:
I was thinking like something in the frontend.

Arsh25, 2016-08-03:
How will the frontend know that the password on the backend is wrong? We would have to set a flag or something that says wring password entered on backend but this means the backend is writing to Superstar without auth.

mrmoss, 2016-08-03:
I re-worded my previous post.

Arsh25, 2016-08-03:
Hmm, I haven't seen that interface. I tried to see it by giving a password for a robot that did not have a password but it just let me through anyway.

mrmoss, 2016-08-03:
I believe stealth_layla has a password.

Arsh25, 2016-08-03:
It does not. The test site has no auth file and the auth file for the prod file has an empty password for all robots

mrmoss, 2016-08-03:
Regular superstar then?

mrmoss, 2016-08-03:
[image]

FineTralfazz, 2016-08-06:
I set the password for stealth_layla and verified that this works.

Arsh25, 2016-08-06:
So I had an auth error and the popup window with just ERROR in red does not help. I have no idea what the error was but sort of guessed it was auth and confirmed that it was an authentication error since the status area showed a message saying check the password. I think the error popup should specify what the error is. 

Did someone work on this? @FineTralfazz  @mrmoss I have no authority but I think this should be reopened

mrmoss, 2016-08-06:
Still open.

Arsh25, 2016-08-06:
If someone is already working on this just unassign and me assign yourself.

FineTralfazz, 2016-08-06:
Sorry, I confused backend and frontend.

Arsh25, 2016-08-24:
@mrmoss I believe you implemented this?

mrmoss, 2016-08-24:
I implemented it...but I was hoping someone would actually design/put some
thought into making it instead of getting the bare bones system to work...
Everything is in popup.js.

Arsh25, 2016-08-24:
By everything you mean 3 lines? What do you mean by bare bones system? Does it still just say ERROR! and not actually give the error? I would test but don't have an Arduino handy.

mrmoss, 2016-08-24:
The 3 lines of code in popup.js and the 16 lines of html in popup.html are all of it. By barebones I mean an error is seen and it pops up a popup.
[image]

Arsh25, 2016-08-24:
Ah I see so it is more of a design thing now instead of functionality.

Arsh25, 2016-08-24:
@robotmoose/development If anyone is interested in making stuff look pretty then this is all yours.