robotmoose/robotmoose #126: Add change password interface to frontend connect modal.

mrmoss, 2016-08-03:
Seems like the most appropriate place to put it.

Need:
Three password type input boxes, first is the original password, second and third are new auth and verification of new auth. Check if last two are the same, if so call superstar.change_auth...

Michaeasmith, 2016-08-03:
Do we want the new auth boxes as: 1) a new modal, 2) hidden/collapsed input boxes that pop up on button press, or 3) input boxes always visible?

mrmoss, 2016-08-03:
I'd say a new modal (just a link on the connect modal to closes the connect modal and opens the auth modal).

mrmoss, 2016-08-03:
Other notable things:

- Interface should only check if the auth change was successful once everything is typed in and the submit button is hit.
- Once the submit button is hit, the values of the three inputs should be copied to local variables and then cleared.
- Valid auth codes are 8 characters and contain no whitespace or non-printable characters (as in [specification](https://docs.google.com/document/d/1oGs5zI263Xecrs5HO5Bf2DIXb9hnaWFIOWkNzhQ9jhI/edit#heading=h.9vf514e7bwp3)).

Michaeasmith, 2016-08-04:
Is there an existing robot with password I can use for testing? 

Or I could create my own. What is the current procedure for creating a new robot and giving it a password?

Arsh25, 2016-08-04:
You can use the starcpy.py util in /utils/superstarv2 to copy a robot. I checked the server and there is no auth file on the test server. I would suggest just making a new robot and setting the auth on the server.

mrmoss, 2016-08-04:
I'd make a robot with an auth code locally and test it (run starcat.py on /
on either server and redirect into db.json).

mrmoss, 2016-08-04:
Alternatively, download the db from the site:
http://test.robotmoose.com/superstar/

commit 3a6dcadc (2016-08-06): Added change authentication, closes #126
closed 2016-08-06 by Michaeasmith