robotmoose/robotmoose #127: Main readme update.

mrmoss, 2016-08-03:
The main readme is really out of date. Should be shot more torwards chromeapp and using the Arduino IDE (since the Makefile upload is mainly for us).

The two sections in need of major re-writing: Installation and Setup and Configure and Pilot your Robot with the Web Front End.

Might be worth touching up the first section as well...

FineTralfazz, 2016-08-04:
I made some updates to it. Feedback?

mrmoss, 2016-08-04:
Looks good to me. Only one small detail, what's the official name for the OS on Macs nowdays?

- OSX?
- OS X?
- Mac OSX?
- Mac OS?
- Mac?

Literally have no idea...

FineTralfazz, 2016-08-05:
[They renamed it to macOS a few months ago](http://www.theverge.com/2016/6/13/11906214/apple-mac-os-10-announced-features-siri-wwdc-2016). Yes, with a lowercase m. Sigh.

mrmoss, 2016-08-05:
Then I'd say you can close this.

closed 2016-08-05 by FineTralfazz