robotmoose/robotmoose #131: C++ Superstar Client Needs Testing

rykerDial, 2016-08-05:
The C++ superstar client is now ready for testing. It has the same interface as the python version, so the unit tests first need to be converted to C++. The superstar source and all its dependencies are in subsystems/kinect_localize, in the include and source files. These files should be in the includes folder, but we now have two different files named string_utils.cpp

mrmoss, 2016-08-05:
The string_util files are probably similar if not subsets of one another
(my standard string functions from msl).

rykerDial, 2016-08-05:
It would be pretty easy to just merge the two. Though I would have to change the .hpp header to .h to make sure I don't break anything.

Arsh25, 2016-08-05:
I don't like where the superstar client is located in terms of directory structure. But I honestly don't know where it should go. Maybe a folder with all the superstar clients? so superstar/clients maybe?

mrmoss, 2016-08-05:
Better would be include/superstar, then rename curremt superstar folders to
superstar_server.

mrmoss, 2016-08-05:
When it comes to the whole rename .h to .hpp, you're going to have the same problem on the other side of things (going to have to change all .hpp to .h).

Grep is your friend in this situation:
```
grep -r string_util.h .
./include/ini.cpp:#include "string_util.h"
./include/robot_config.cpp:#include "string_util.h"
./include/string_util.cpp:#include "string_util.h"
./tabula_rasa/backend.cpp:#include "../include/string_util.h"
```
Replace all of the above with .hpp.

rykerDial, 2016-08-05:
I would like it if we split up our header and source files into include and src directories. I think that would make it a bit cleaner.

rykerDial, 2016-08-05:
I moved the superstar files to include/superstar. Hopefully that makes it easier for you Arsh.

mrmoss, 2016-08-05:
For our stuff, I'd still say put all superstar stuff in /include/superstar/, since then it's fairly simple to compile things:
```
INC_FOLDER=../../include
SS_FOLDER=$(INC_FOLDER)/superstar
FLAGS=-I $(SS_FOLDER)
SS_SRC=$(SS_FOLDER)/superstar.cpp 
```
I don't think it's a good idea to separate mongoose and jsoncpp into different folders (since those are directly from the git repo, so updating means we just put new files into them). And to continue with the previous example:
```
INC_FOLDER=../../include
SS_FOLDER=$(INC_FOLDER)/superstar
MG_FOLDER=$(INC_FOLDER)/mongoose
JSONCPP_FOLDER=$(INC_FOLDER)/jsoncpp
FLAGS=-I $(SS_FOLDER)
SS_SRC=$(SS_FOLDER)/superstar.cpp 
```

rykerDial, 2016-08-05:
Well, it's not that big a deal if they're together. But cmake makes it easy to keep them separate :).

mrmoss, 2016-08-05:
```
import github
repo=github.fetch("github.com/robotmoose/robotmoose")
for issue in repo.issues:
    for comment in issue.comments:
        comment=comment.lower()
        if "cmake" in comment and ("good" in comment or ";)" in comment or ":)" in comment or "not terrible" in comment):
            comment.user.set_permissions("ro")
```

Arsh25, 2016-08-05:
I had to look it up and see if there was a python module for github. 

https://github.com/PyGithub/PyGithub

rykerDial, 2016-08-05:
Haha. But honestly what makes cmake so terrible?

mrmoss, 2016-08-05:
It seems like a waste. Cmake makes Makefiles, why not have a new program called Makemake that makes Cmake files which you can then use to make Makefiles that you can then use to make a Makefile?

Also, I've grown to hate it by using it. If you keep getting beat with a sack of oranges...you might start to hate sacks and/or oranges...

Arsh25, 2016-08-05:
I think the helper functions (eg stringify) should not be aprt of the interface. Clients should not have any reason to use these fubtions, the only thing that should be exposed are the methods specified in the spec (get,set,get_next etc)

mrmoss, 2016-08-05:
They aren't, you only include superstar.hpp when using the client interface, so you don't even see them, but you need them in includes to be able to compile things.

rykerDial, 2016-08-05:
I don't remember stringify, but functions like build_skeleton_request could
be made private.

mrmoss, 2016-08-05:
Sounds good.

Arsh25, 2016-08-05:
Wrong example, I meant to say functions like pathify,strip_trailing_slashes etc.

mrmoss, 2016-08-05:
Other than the cmake stuff, code looks good Ryker. I agree the helper functions should be made private.

rykerDial, 2016-08-12:
Tested set, flush, auth, and callbacks