robotmoose/robotmoose #133: Soft wipe chromebooks in the classroom

olawlor, 2016-08-05:
It'd be nice to clear students' browser history (and login info) when our chromebooks are booted or woken from sleep.

computations, 2016-08-10:
I did some research into this. Google Chrome stores all of its user data in ~/.config/google-chrome/Default/, and its safe to just rm -rf that directory to clear history/logins/saved passwords (chrome will just remake everything that needs to be remade).

Secondly, most distros of linux use anacron as well as cron. anacron is a anachronistic version of cron, that makes sure cron jobs are run close to when they need to. If we wanted to add a daily job to clear history, we can just add a script to the /etc/cron.daily/ directory. (see https://www.digitalocean.com/community/tutorials/how-to-schedule-routine-tasks-with-cron-and-anacron-on-a-vps) for more info.

mrmoss, 2016-08-10:
I was messing around, and I came up with the below bash script. Seems to not keep history and cookies on closure of chrome while preserving localstorage and chromeapp data:
```
#!/bin/bash
chrome_dir=~/.config/google-chrome/Default
for ii in Last\ Session Last\ Tabs Cookies Cookies-journal History History-journal
do 
    rm -f "$chrome_dir/$ii"
    ln -s /dev/null "$chrome_dir/$ii"
done
chrome_bin=$(which google-chrome)
if [ ! -f $chrome_bin.old ]
then
    sudo mv $chrome_bin $chrome_bin.old
    echo -e "#!$(which bash)\n$chrome_bin.old --noerrdialogs"|sudo tee $chrome_bin > /dev/null
    sudo chmod +x $chrome_bin
fi
```

computations, 2016-08-10:
One note, cron probably doesn't run as the user, so ~ points to the wrong directory. That should be expanded to /home/$user. Also, we might want to add `Login

computations, 2016-08-11:
You know, if we are going to wrap the executable in a script, then why not just use the --incognito switch and not worry about linking stuff to null.

mrmoss, 2016-08-11:
It's easy to get out of incognito, in which case we have the same problem
(file>new window and ctrl+n will open in normal mode).