robotmoose/robotmoose #135: Superstarv2 auth changes.

mrmoss, 2016-08-06:
Enable all robots to have auth set (even blank passwords.
Add special value to disable writing auth.

mrmoss, 2016-08-09:
This is done...I think...could I get some external testing verification (up on test.robotmoose.com)?
@FineTralfazz @arcshock @computations @Arsh25 @rykerDial @olawlor @Michaeasmith 

Current auth file:
```
/ admin
/robots/2016/uaf schoolpass
/robots/2016/uaf/stealth_layla !
/robots/2016/uaf/azure_layla 123
/robots/2016/uaf/bot 456
```

If multiple people test on the test server, I'm imagining everyone will be overwriting each other...so it might be better to test locally.

No documentation yet...but currently:
- Having a "!" in auth file means a path's auth is immutable. You can't make an auth code for it or any its children, or children's children, etc...UNLESS you use an auth code above that directory that isn't immutable and none of that auth code's parents are immutable.
- Immutable auth directories can be written to.
- Blank auth is not the same as not having a password.
- Not having a password means use your parent's password.
- Having a blank password means the password is blank.

One small problem with this whole auth implementation:
Sub levels with auth codes that are the same as their parent's auth code will take the auth code of the top-most parent...not sure whether this is good or bad...

@Michaeasmith I notice I seem to keep breaking the sim code on the frontend...I apologize...

Michaeasmith, 2016-08-09:
Encountered an error when testing locally: When I try to change authentication of a robot with no password, the error pops up twice: "Superstar error(0) - NotFoundError: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node." 
[image]

This seems to happen whether I leave the Old Authentication field blank or not.

Michaeasmith, 2016-08-09:
I should add, my auth file consists only of the line:

/robots/2016/uaf/test password

mrmoss, 2016-08-09:
That's weird...that's a javascript error in modal_change_auth.js starting at 285. It shouldn't be able to error, since it only removes nodes that already exist...

What if we just wrap that in a try catch and ignore any errors in there?

mrmoss, 2016-08-09:
Was your superstar server running? The page may have been cached. Error 0 means the error happened before any networking operations (or that it couldn't reach the server, HTTP error 0).

Michaeasmith, 2016-08-09:
Oh my bad, I think I was still running the old superstarv2 before the recent commit. No longer getting that error after restarting superstar.

Michaeasmith, 2016-08-09:
It seems to be working locally as you described in my tests so far. But it seems odd to me that a robot with no auth is not the same as a blank auth.

So with an auth of:
/robots/2016/uaf/test password

It is not possible to connect to any robot except for 2016/uaf/test - every other robot gives an "incorrect auth" error.

I suppose this will not be a problem as long as when we make a create robot utility we have it always add the robot to the auth file (with an auth of " ", "!", or a password)

Arsh25, 2016-08-09:
There is an easy fix to this. For testing you could just make / have an empty password that would allow all robots with no entries in the auth file to have an empty password. O don't know if this an implementation limitation or on purpose but for production all robots will have an entry in the auth file and whatever way we create to make robots must take this onto account

Michaeasmith, 2016-08-09:
If you add / with an empty password to the auth file, then it seems the authentication field is ignored entirely for all robots. You can put whatever you want there. So this wouldn't help for testing passwords. Is that the expected behavior?

Arsh25, 2016-08-09:
I'm going to call that an edge case that probablywasnt tested. I think it's a side effect of allowing the parent auth to work on the child so since / is empty an empty password will work on all robots but an incorrect non empty password should not.

mrmoss, 2016-08-09:
Yes, that means the root password is blank.

mrmoss, 2016-08-09:
Just changed the auth system to use hashes instead of plaintext auths. This more than likely breaks you local setup (you'll need to change your auth file to use hashes).

You can generate auth codes in two ways:
Anything but windows (unless you use gitbash/cygwin) and macOS: `printf authcode|sha256sum`
macOS: `printf authcode|gsha256sum`
Anything: [http://www.xorbin.com/tools/sha256-hash-calculator](http://www.xorbin.com/tools/sha256-hash-calculator)

Note: "!" still means immutable password and is not hashed.

Test server auth file was updated to using hashes.

Michaeasmith, 2016-08-10:
Testing locally, it seems that if you have a blank line in the auth file, then any robot is accessible without authentication. Is that expected behavior?

mrmoss, 2016-08-10:
Yes...and no...

Pathification replaces all consecutive slashes with a single slash and then
removes all trailing and leading slashes.

So "/" gets turned into "", and since there's nothing next to it, the auth
is technically empty for it. So technically, an empty line is the same as
giving a blank auth to /.

I guess that shouldn't be a thing...just pushed a fix.

closed 2016-08-12 by mrmoss