robotmoose/robotmoose #137: Kinect data disappears from sensor data object

rykerDial, 2016-08-11:
Whenever superstar receives sensor data from the Arduino, it replaces the old JSON. If the Kinect data is stored in robot.sensors, this overwrites it. This is expected behavior.

There are a few work-arounds to this:
1. Separate robot.kinect. Data would persist there.
2. Separate robot.kinect, and robot.kinect is put inside robot.sensors before the sensor json is sent to the various doorways. This is nice because the angle data also then shows up in the sensor tab, but it just feels kind of wrong to duplicate the data in this manner.

Any thoughts?

Arsh25, 2016-08-11:
My vote goes for option 2. Here are my reasons:

1 Putting Kinect data into robot.sensors is ok for now when all the Kinect is reporting is 1 float but we may want to add more stuff from the Kinect in the future and having a separate Kinect object created only when a Kinect is connected is better for when we have more stuff from the Kinect. 

2 Ownership. I think only the Arduino should be responsible for setting the sensors object. This prevent unnecessary copying and also keeps diagnostics easy if sensor information is incorrect. 

3 Kinect data should still show up under sensors if it exists. The same ting can be done for charts, you add a Kinect chart and it charts all Kinect data?

rykerDial, 2016-08-11:
Good point. Since angle data is only one that can be displayed in sensor tab, we'll merge angle only. 

In the current setup, charting kinect would add all, but it can be easily changed so we can choose individually.

Arsh25, 2016-08-11:
To clarify, I think that all Kinect data should be separate. So /kinect would look something like 
`kinect:{"angle":45.56}`.In the future this could be expanded to something like `kinect:{"angle":45.56,"depth":[34,67,45]}` etc

mrmoss, 2016-08-11:
I think the easier route is to change the backend from writing the entire
sensors object to only writing values within it. This way the kinect can
write data into sensors as well...

rykerDial, 2016-08-11:
Yeah, kinect data is already in the format 
`kinect {"angle":value}`

rykerDial, 2016-08-11:
Changing the backend to only changing values within the sensors object is probably a good idea regardless, but we still wouldn't want depth data going to the sensors tab. The two ways to do that are to either pass over that data in treeviewer or have a separate object that stores the big data.

mrmoss, 2016-08-11:
I'm not a big fan of having the object in the path twice. This is literally a one line fix in the backend...could I get someone to check my last commit?

mrmoss, 2016-08-11:
I'm ok with having the array in the sensors tab...isn't it hidden until you click the + next to it? We can even have a custom view in treeviewer to show an image.

rykerDial, 2016-08-11:
What happens now when we add a sensor and then remove that sensor? Since the sensor object is no longer replaced, does that mean that the last value for the sensor data will persist forever?

mrmoss, 2016-08-11:
Good catch, leave it to the person who puts in that data to clean it up,
just pushed.

mrmoss, 2016-08-11:
This obviously doesn't address the Kinect data. There's no configuration option for it, if there was, the backend would already clean it up if "kinect" was added to the sensor options object in connection.js.

Before we go on any further on this, it is probably worth waiting for a decision from @olawlor.

rykerDial, 2016-08-11:
Alright, sounds good. I could have the kinect backend send of set request when it exits, setting kinect to null. This would work unless superstar was shut down before the kinect backend.

mrmoss, 2016-08-11:
That'd work.

Questions for Friday's meeting/whenever:
- Old sensors...do we care about them?
- Should Kinect just be it's own thing outside of sensors?
- Is it worth looking/have we looked into using something like [DepthJS](https://github.com/doug/depthjs) to have the Kinect talk directly to the backend?

mrmoss, 2016-08-11:
Location has the same issue the Kinect does, even more so since multi sources can edit it (slam and lidar/webcam and opencv/encodes/frontend).

I guess there could be a sub sensors value in location that stores the raw values given from each of the different sources, and the main location you see is the fused sensor reading...

Another question for Friday I guess...

olawlor, 2016-08-11:
We can discuss this further Friday, but the main conflict here seems to be between these two design goals:

1. For the backends, it should be easy to add new strange devices, with their own separate way to access superstar.  This argues for separation between device data, for example by splitting robot/sensors (all Arduino-derived data) from robot/kinect (all Kinect-derived data).
2. For the front end, it should be easy to see and access all the robot data from all the devices.  This argues for presenting everything in sensors, such as sensors.kinect.

So far, we've heard these solutions:

- (A)rsh: use (1) for the backend but fake (2) on the frontend by manually merging the objects.  I like the simplicity of this, although it does make it hard for new people to add new device types.
- (M)ike: support (1) by writing each sensor value separately, so the frontend can just grab all sensors.  I like that people can add new devices easily, although I'm concerned about the overhead of a dozen separate writes (even if jsonrpc allows them to be batched).

Both solutions have a problem with stale data; "sender cleanup on exit" sounds good but doesn't work if the sender crashes or loses network.  The only reliable solution I know of is heartbeats.  For solution (A), that's 2 heartbeat values, one of which already exists.  For solution (M), that's a separate heartbeat for each sensor, adding more overhead.

A possible hybrid solution would be to move the current robot/sensors backend data in superstar into robot/sensor/arduino, which can be written in one big chunk, and other devices go in robot/sensor/kinect or robot/sensor/lidar.  The frontend could then grab the whole robot/sensor object, and automatically merge the up-to-date values (heartbeat updating) for the presentation layer.  This would break backward compatibility, although you could have a fallback by keeping the old robot/sensors path.

mrmoss, 2016-08-11:
I'm starting to think the merging+heartbeats is the most elegant solution... This whole cleanup thing didn't occur to me, and it is a big extra step.

As for changing the sensors layout, I'm not sure if it is necessary, but we've already broken backwards compatibility with superstarv2...so breaking it further isn't the most awful thing in the world.

mrmoss, 2016-08-11:
What if we implemented a timeout on the frontend?

Every time the frontend receives a new kinect packet with getnext, reset the timeout. If the timeout reaches zero, remove the local sensors object. This would remove the need for a heartbeat.

commit efb032d1 (2016-08-11): Angle data is now merged with sensor data object. Closes #137