robotmoose/robotmoose #138: Sim sensors not updating.

rykerDial, 2016-08-12:
When you simulate a robot, only heartbeats changes in the sensor object. With a create 2 configured, you can see encoders updating in the map but not in the tree-viewer.

Michaeasmith, 2016-08-12:
I am not able to reproduce this bug. Sensors are updating normally for me both on dev site and local. Can you elaborate your setup? Dev site or local superstar? Any devices other than create2 configured? Is your cache clear?

mrmoss, 2016-08-12:
Location is updating, just encoders.

Michaeasmith, 2016-08-12:
There is an issue for encoders on the backend, but not for the frontend simulator, that I know of. Should be easy to add. In a real robot is the value simply the distance each wheel has traveled in meters?

mrmoss, 2016-08-12:
I believe it's the number of encoder ticks, not meters.