robotmoose/robotmoose #141: Add laptop battery level to backend.

mrmoss, 2016-08-12:


rykerDial, 2016-08-13:
This is actually pretty easy to implement, it is just a question of where we want it in the files. Should we add this to connection.js and do a superstar set of the battery info at the same time we set the sensors?

mrmoss, 2016-08-13:
My vote is for un the sensors object.

Arsh25, 2016-08-13:
Something amazing is going to happen. I agree with Mike it should be added to sensors object. Maybe something like laptop_battery and then you can have a charge and charging state

rykerDial, 2016-08-13:
Just wondering if it logically makes sense to have a battery_status_t be a member of connection_t since connection.js deals with arduino data primarily

Arsh25, 2016-08-13:
It doesn't but that's good for now. Or we could hold off until the refactor that puts Arduino stuff into sensors/Arduino and then put laptop_battery understand sensors. You know do it right the first time

rykerDial, 2016-08-13:
This question isn't about where it should post to vut about where the code should live in the backend files. Since connection.js already handles all the superstar comm and synchronization and stuff, there would be a lot less duplication if we have connection.js be responsible for the battery object.

Arsh25, 2016-08-13:
Ah yeah I know think connection.js is ok.

mrmoss, 2016-08-13:
I'd just add it into connection.js where ever they sensors object is
created and sent.

rykerDial, 2016-08-13:
Alright, backend now merges laptop battery information with the sensors object before it pushes to superstar. This could use some testing on other systems, but as long as chrome is at least version 39 it should work.

rykerDial, 2016-08-15:
Works on backend computers.

closed 2016-08-15 by rykerDial