robotmoose/robotmoose #144: Back and frontEnd gruvei call names do not match

Arsh25, 2016-08-19:
The back and frontend are creating different hashes for the call name. This was encountered on an updated repo.

mrmoss, 2016-08-19:
> Assigned #144 <https://github.com/robotmoose/robotmoose/issues/144> to
> @mrmoss <https://github.com/mrmoss>.

Are you using the same auth code? Are you using thr first pilot to connect?

Arsh25, 2016-08-19:
There was only 1 pilot and both sides where using an empty auth code. piloting worked so with code was correct.

mrmoss, 2016-08-19:
Just because the auth input on the backend is empty it doesn't mean the
auth is empty (it's cleared when you hit connect). As I recall, the auth
code is saved as well. Was it using a previous auth code?

Arsh25, 2016-08-19:
No auth was previously used. I don't believe if we save the auth code, if we do and it shows an empty auth field then that should be filed as a bug.

mrmoss, 2016-08-19:
I can't reproduce this (unless I use a pilot that is not the first pilot)...so I just removed the single pilot based video altogether...

Arsh25, 2016-08-19:
What do you mean by first pilot? The first pilot to ever connect to the robot or the first pilot amongst the current list of pilots? If it's the latter than we just had one pilot while testing but there were older pilots who were no longer connected and did not show up on superstar too

mrmoss, 2016-08-19:
The pilot with the lowest uuid in the current list under frontend_status (the oldest pilot that is connected).

mrmoss, 2016-08-19:
Was someone else connected? Like right now, someone other than me was connected to stealth_layla, so I had to find a robot with no pilots to test with...

Arsh25, 2016-08-19:
Ok. So we had it not work when there was only one pilot connected

Arsh25, 2016-08-19:
Nope, the backend reported only 1 pilot connected when we had the issue

mrmoss, 2016-08-19:
Weird...I tried for a good 20 minutes to reproduce this. Did you have an old version of the backend?

Arsh25, 2016-08-19:
No, I pulled the repo before using the backend

mrmoss, 2016-11-04:
Did you check your app version?

olawlor, 2016-11-04:
If you open a new backend and connect to a LAYLA with an auth code, the backend's gruveo link is wrong until you disconnect and reconnect.  This is with 2016-10-25 roadkill backend, and the production robotmoose.com frontend.

mrmoss, 2016-11-05:
Alright, I think it's working. I removed the whole auth clearing bit of
code (changes the input box...thus why the recall was commented out in the
onchange event).

It looks like production apps have the inspector restricted, so you cannot
just "look" at an auth code with right-click+inspect...

closed 2016-11-05 by mrmoss