robotmoose/robotmoose #147: HTTPS introduces substantial delay

rykerDial, 2016-08-23:
Using http with the Kinect, data posts to superstar just fine and few packets are lost. Using https, many packets are lost, substantially reducing the kinect data update rate.

Arsh25, 2016-08-24:
Are you using KEEP-ALIVE? I can totally see the lag being introduced due to each update having to do the TLS handshake.

mrmoss, 2016-08-24:
It is not krep alive (single request via mongoose). Mongoose does have the
option, but this would be a major refactor to wget.hpp/wget.cpp.

rykerDial, 2016-08-25:
We should use keepalive if we ever expect a superstar client to need to use https on a regular basis. But with the way superstar is setup now, everything posted to it is publicly viewable anyway, so clients should not be posting sensitive information.

Arsh25, 2016-08-25:
Yeah I agree we aren't really posting any sensitive information. The only problem with using HTTP is the possibility of a replay attack but honesty there's not much harm that can be done apart from posting bad data to superstar

Arsh25, 2016-08-25:
I think we should use keep-alive anyways because that could help minimize the latency (though there is no way to tell without experimentation) but that is a big change and should  be discussed and/or its own issue

mrmoss, 2016-09-24:
This isn't https, at home all kinect data is bad for me (this is latency period, https just happens to be enough on campus to cause the error, but http causes the error at home as well).

closed 2016-09-24 by mrmoss