robotmoose/robotmoose #148: Pilot video code in use

Arsh25, 2016-08-24:
I believe this is because a video code is only generated for the first pilot and any subsequent pilots use the same code? Also when does a pilot get removed from superstar? If they don't then they should after some time of inactivity. 

[image]

[image]

mrmoss, 2016-08-24:
> Assigned #148 <https://github.com/robotmoose/robotmoose/issues/148> to
> @mrmoss <https://github.com/mrmoss>.

I removed multiple pilots, there is only one pilot now (thr count is there,
but the video code is the same for every pilot).

Arsh25, 2016-08-24:
Yeah I saw that so it just uses the robot name and auth to create the code. My guess is that someone is on the stealth_layla video but I don't see any other pilots connected.

mrmoss, 2016-08-24:
Are you doing things with it right now? I can try clearing the pilots.

Arsh25, 2016-08-24:
Nope not doing anything right now since I can't get the video to work.

Arsh25, 2016-08-24:
The issue is with that code being used by 2 people already. I think it's some sort of incorrect hangup that gruveo did not receive properly or a rogue laptop connected to the video but that should show up as an active pilot (if connected via robotmoose). I don't know how to solve this since the code is based on robot name and auth apart from changing the auth.

mrmoss, 2016-08-24:
When's the demo? I can run into Chapman now (might be an hour or so).

Arsh25, 2016-08-24:
It's just to Seta, an hour or so would be ok

mrmoss, 2016-08-24:
This is weird, the robot was even saying the code was in use and constantly
reconnecting... Reconnect with the root pass and it should work.

Arsh25, 2016-08-24:
Thanks for looking into this so early. I'll give it a try

mrmoss, 2016-08-24:
How's it going? Should we connect to a gmail chat or something if we need
it?

Arsh25, 2016-08-24:
> auth/ABvm0-gwLM2IZn-

So that worked. I may have been running a rogue robot so please check if
it's on the charger. I tried driving it and either it wasn't moving or the
lag was crazy. I could see heartbeats so if it hasn't moved then the power
is probably off.

mrmoss, 2016-08-24:
I was there, it wasn't driving at all. I just checked it, looks fine...I
can drive it...

Arsh25, 2016-08-24:
Probably the lag then. It was working so well last night. Anyways there's
nothing more scheduled for today. May run it Friday again

rykerDial, 2016-08-25:
I'm having similar issues in Sitka, using my phone as a hotspot. I can see the sensor updates, see the video stream, and play sounds, but can't drive.

mrmoss, 2016-08-25:
I'll check on it when I get in. Might be comet not re-callong its-self due
to some weird edge case.

mrmoss, 2016-08-25:
Alright, I think I figured out the problems with this. There's a
network_getnext in connection.js that sets the getnexts for pilot and
config. On error they weren't retrying (robotmoose went down last night?!).

Fix is up on git, soon to be on the store.

closed 2016-08-25 by mrmoss